roboticslab-uc3m/openrave-yarp-plugins #5: Update README with block diagram

RaulFdzbis, 2016-11-11:
Update README with block diagram for a further explanation of the system.

RaulFdzbis, 2016-11-17:
Block diagram added in e577f61b2383788b666c704216cd6051b8e46ae5 . Some explanation should be also added

RaulFdzbis, 2016-11-18:
This can be considered closed with 3b0c5ca78aad76c70776f7eb01dc4e152702771d.

jgvictores, 2016-11-19:
Reopening, not sure if diagram reflects reality. E.g. the robot is optional, and is through the yarp plugin, not directly to the openrave component. Additionally, naming should match the folder structure (openraveplugins and yarpplugins).

RaulFdzbis, 2016-11-21:
The naming was choose to match the executable of the module. The problem i see with using the folder structure as the scheme is that for example the Openrave-YarpRobot module make use of the two different folders (openraveplugins and yarplugins). 

Maybe the problem is that the diagram reflects a specific application of the openrave plugin, which is using the robot. Maybe if we put it in the "How to use the openrave-yarp-plugins for collision avoidance with teoSim" tutorial would make more sense. Then the "general" scheme should focus on how the plugin itself works.

RaulFdzbis, 2016-11-21:
After some working with @jgvictores the diagram was updated. The descriptions of the different modules were deleted, in order to not lose generalization.

closed 2016-11-21 by RaulFdzbis