roboticslab-uc3m/openrave-yarp-plugins #9: Returns remotes to the local plugin

RaulFdzbis, 2016-11-15:
It would be interesting to return the results of some of the remote functions declared in the remote robot to the local plugin, in order to be used in the local scope.

RaulFdzbis, 2016-11-16:
In 676c796e039d808845c46745cd6ef2036614f405 some work has been done to return remotes functions of the "get" type to the local plugin. @jgvictores are you interested in returning more functions? or we can mark this issue as closed?

jgvictores, 2016-11-16:
@RaulFdzbis Could you see my comment on the commit ( https://github.com/roboticslab-uc3m/openrave-yarp-plugins/commit/676c796e039d808845c46745cd6ef2036614f405 ) (not sure if a notification reaches you upon commenting on the issue)

RaulFdzbis, 2016-11-16:
I received the notification. I will fix it as soon as possible.

jgvictores, 2016-11-19:
Happy with https://github.com/roboticslab-uc3m/openrave-yarp-plugins/commit/dbc459bd9b1ba290c50a11630a58afa92a3a417c, closing issue!

closed 2016-11-19 by jgvictores