roboticslab-uc3m/openrave-yarp-plugins #20: teoSim conf wrong installation path

jgvictores, 2017-03-20:
teoSim conf wrong installation path.

depecated teoSim means not to prioritize, but may affect all `share`.

jgvictores, 2017-04-02:
Fixed at e755e0b77126cfc5f5ae64ad8eb72005c98478f6 by reordering root cmake, but detected very strange effect that maybe should be reported upstream (YARP): `yarp_configure_external_installation` helpers are broken if `include(YarpPlugin)` is called previously.

PeterBowman, 2017-04-02:
It looks like this nails down to the `CMAKE_INSTALL_DATADIR` being set to the empty string by the [GNUInstallDirs CMake module](https://github.com/Kitware/CMake/blob/master/Modules/GNUInstallDirs.cmake) (not sure yet how nor why, note that CMake cache is extensively used there) on its first run in previous subdirectories. The value of this variable (`share`) should be prepended to `${ROBOTICSLAB-OPENRAVE-YARP-PLUGINS_CONTEXTS_INSTALL_DIR}`; we get a string introduced by a leading slash otherwise, hence the absolute path.

As an alternative solution to yours, you could have typed the following line before the first call to `yarp_configure_external_installation()`:

```cmake
include(GNUInstallDirs)
```

It should be noted that `YarpPlugin` already [includes](https://github.com/robotology/yarp/blob/cca890b/cmake/YarpPlugin.cmake#L53) the `YarpInstallationHelpers` module. Then, when we want to enter in `include(YarpInstallationHelpers)`, [an if-guard](https://github.com/robotology/yarp/blob/cca890b/cmake/YarpInstallationHelpers.cmake#L7) prevents us from filling/updating the values of `CMAKE_INSTALL_*` variables by a `include(GNUInstallDirs)` command in the next few lines.

PeterBowman, 2017-04-02:
Oh, well. The `CMAKE_INSTALL_*` variables are meant to be set to a null string in the cache so that the user may find it and specify an appropriate value by means of a graphic manager or similar ([proof@v3.5.2](https://github.com/Kitware/CMake/blob/v3.5.2/Modules/GNUInstallDirs.cmake#L246)). The actual default value (`share`) is set **locally**, that is, it's only visible in the scope where `include(GNUInstallDirs)` is called.

PeterBowman, 2017-05-05:
The upstream patch was merged at robotology/yarp@229ca58, it's targeted for YARP release 2.3.68.1.

jgvictores, 2017-05-06:
Closing as working locally thanks to e755e0b77126cfc5f5ae64ad8eb72005c98478f6, and fix merged upstream. Thank you so much!

closed 2017-05-06 by jgvictores