roboticslab-uc3m/openrave-yarp-plugins #25: Compute YCM-aware relative path to OpenRAVE plugins dir on installation

PeterBowman, 2017-04-20:
See c7284d1 and its commit comments (pasted below). This issue affects YCM projects depending on openrave-yarp-plugins, currently: [teo-main](https://github.com/roboticslab-uc3m/teo-main/) and [asibot-main](https://github.com/roboticslab-uc3m/asibot-main/) (WIP).

> PeterBowman commented on c7284d1 on 16 Mar
>
> @jgvictores this patch was meant to provide a relative installation path during the YCM build in teo-main. However, YCM keeps spoiling `CMAKE_INSTALL_PREFIX` with the local `build/install/` path. I can't find a satisfactory solution, apart from instructing users to perform a `sudo make` on each build. We'd need a constant CMake variable always set to `/usr/local` (or the platform equivalent).

> jgvictores replied on 16 Mar
>
> Not sure if I've fully understood the problem and/or if the following could be a... reasonable hack:
> * Check if `CMAKE_INSTALL_PREFIX` is a substring of `OPENRAVE_PLUGINS_INSTALL_DIR`.
> * If so, remove it!
> What would your opinion be on implementing this?

> PeterBowman replied on 16 Mar
>
> There are two relevant use cases, let's assume `OPENRAVE_PLUGINS_INSTALL_DIR` points at `/usr/local/lib/openrave9-plugins`:
> * openrave-yarp-plugins as a standalone project (no superbuild involved) and default/unset `CMAKE_INSTALL_PREFIX` (`/usr/local`): both methods would work, yours and the relative path way proposed in this commit
> * openrave-yarp-plugins as an external component downloaded on demand (e.g. in teo-main): here, YCM sets `CMAKE_INSTALL_PREFIX` to, say, `/home/username/repos/openrave-yarp-plugins/build/install`
>
> We can't extract `CMAKE_INSTALL_PREFIX` as a prefix in the second case. If there was a CMake variable/property which stored the default value (`/usr/local`), we could then resort to plain string manipulation.
>
> I'm actually questioning the purpose of this commit: it makes no difference whether we pass an absolute or relative path to install() unless we find a way to properly propagate the latter behavior to superbuild-like projects.

> jgvictores replied on 16 Mar
>
> Oh, ok. Got it now.

PeterBowman, 2017-04-21:
Should work as expected at 67714f9. Now, there are two situations that may arise here:

1. the user compiled OpenRAVE from sources and installed it in a custom path (different from `/usr/local`)
2. the user supplied a custom `CMAKE_INSTALL_PREFIX` when compiling this repo

Both circumstances would result in copying OpenRAVE plugins to the wrong path upon install. However, similar issues could have also derived from handling YARP plugins and we didn't bother to address them in our code (variables like `YARP_DYNAMIC_PLUGINS_INSTALL_DIR` are always relative to the installation prefix; since we don't use their absolute-path counterparts, one could assume that finding a proper solution here was not a compelling matter).

Closing, please reopen if any of the points above should be reconsidered.

closed 2017-04-21 by PeterBowman

jgvictores, 2017-04-22:
@PeterBowman Great, thank you so much!