roboticslab-uc3m/openrave-yarp-plugins #27: Life beyond teoSim

jgvictores, 2017-05-03:
What we have in this repo is:
- A YARP Controlboard for each robot, which does not emulate the velocity of the robot joints (provides instant movements). Loaded by `OpenraveYarpControlboard`, called `YarpOpenraveControlboard`.
- A YARP Controlboard for each robot, which does not emulate the velocity of the robot joints (provides instant movements), does not allow collision, and can pass movement to another robot (e.g. the real robot). Loaded by `OpenraveYarpControlboard`, called `YarpOpenraveControlboardCollision`.
- OpenraveYarpPaintSquares, used for XGNITIVE painting demos.

Each of the following RECOVER could/should be implemented as separate openrave plugins (with yarp plugin if required).

teoSim offered (most code in the [`run()`](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/479b52dcebf10eab1b75f9f2afe05d544f81d095/programs/teoSim/TeoSimRateThread.cpp#L33-L148) function):
- A YARP Controlboard for each robot, which emulated the velocity of the robot joints. RECOVER :truck:
- Publishing each rgb and depth camera via YARP. RECOVER :truck:
- Publishing each force/torque sensor (requires plugin) via YARP. RECOVER :truck:
- Loading the `ExternObj` plugin. We can do this from python now. FORGET this functionality, but RECOVER ExternObj from asibot-main.

Regarding [src/libraries/RlPlugins/ravebot](https://github.com/roboticslab-uc3m/asibot-main/tree/ec0b387fe3ea7eed6ccd0087f296dd0172193730/src/libraries/RlPlugins/ravebot),
- A YARP Controlboard for ASIBOT, which emulated the velocity of the robot joints. WONT-FIX (DUPLICATE).
- WorldRpcResponder. RECOVER :truck:
- MobileRpcResponder. Maybe RECOVER :truck:
- ExtraCallback. Have no idea.

jgvictores, 2017-05-06:
Related to #26 (Import OpenRAVE libraries and programs from asibot-main repo).

PeterBowman, 2017-05-17:
The following line was added to the description:

> Additionally, [`FakeControlboard`](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/tree/5483749/yarpplugins/FakeControlboard) should be transferred to [yarp-devices](https://github.com/roboticslab-uc3m/yarp-devices) after finally having removed `teoSim`.

rsantos88, 2017-05-25:
I'm working in the first point in this issue (#29)

> * YARP Controlboard for each robot, which emulated the velocity of the robot joints. RECOVER :truck:

rsantos88, 2017-06-19:
I've working in this point:

> WorldRpcResponder. RECOVER :truck:

And now, we can use this as separate openrave plugins. In this moment It's possible and functional to create static objects in the same environment. 
Pending tasks: make functional the `grab` command...
If you want, you can check the example [YarpRobotWorldRpcResponder.py](https://github.com/rsantos88/openrave-yarp-plugins/blob/worldRpcResponder/examples/openrave-YarpRobotWorldRpcResponder.py) :)

rsantos88, 2017-06-22:
Continuing with **WorldRpcResponder**:
* I added a new `info` command to know the robots information (robot and manipulators names)  and the list of all the bodies in the environment (including the robots)
* Now `world grab (manipulator) (obj) (num) 0/1` command is working fine
* Now `world whereis tcp (manipulator)` command is working fine
* it's has been deleted `world mk mesh` and `world draw 0/1 (radius r g b)` commands because I think (maybe) are useless, at least with teoSim.

Note: however, `world whereis obj (name)` is not working well , it shows all the time (0.0, 0.0, 0.0) for all objects :(

rsantos88, 2017-06-23:
More changes with **WorldRpcResponder**:
* Recovered `world mk mesh` and `world draw 0/1 (radius r g b)` at the request of @jgvictores. The firts will make a default mesh (always the same... probably it hasn't been finished the developer intention because you can't enter aditional params). The second I don't know what it can be used for.
* `world whereis obj (name)` command it's working fine. It's just only that this command returns the object position respect to the first position that you make the object. Always that you create an object, the position returned will be (0.0 0.0 0.0) but, if you move the object, this command will return the new position. So, you can make the object always in zero position and then, move the object with `world mv (name) (three params for pos)` to the new position.
* formatted code to make it easier to understand

rsantos88, 2017-06-26:
Pull request with this changes at #31

jgvictores, 2017-11-29:
Now that #29 is closed, we can definitely say there is life beyond `teoSim`.

The future of `teoSim` can be a simple alias of openrave with many plugins and parameters, or a python script that does the same.

rsantos88, 2017-11-30:
:smile: :clap::clap::clap::clap::clap::smiley:

jgvictores, 2017-11-30:
In fact, the current [openrave-YarpRobot.py](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/d433cc5d05455fb12c20ef226f72029f04be73af/examples/openrave-YarpRobot.py) is itself already a good replacement `teoSim`: it loads the default OpenRAVE TEO model and provides movement of each limb to be controlled via YARP `remote_controlboard`.

The script can be easily adapted to incorporate any other of the already developed plugins available in this repository, such as [these](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/tree/d433cc5d05455fb12c20ef226f72029f04be73af/libraries/OpenravePlugins), such as the WorldRpcResponder (script already implemented [here](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/d433cc5d05455fb12c20ef226f72029f04be73af/examples/openrave-YarpRobotWorldRpcResponder.py)).

jgvictores, 2017-12-07:
While could be implemented via `OpenraveYarpControlboard`, best wait to see outcome of `OpenraveYarpPluginLoader` from #47, specifically the multiple device load commented [here](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/47#issuecomment-349556072).

PeterBowman, 2017-12-17:
Please drop me a line at #54 when everything is set.

jgvictores, 2017-12-17:
My personal agenda with `roboticslab-uc3m` is #36 and then #47, which means we will very soon have a worthy successor of the current `teoSim` implementation.

jgvictores, 2017-12-26:
Announcement: #47 has been closed.

We could now map `teoSim` to:
```bash
openrave /usr/local/share/teo-openrave-models/contexts/openrave/teo/teo.robot.xml --module OpenraveYarpPluginLoader "open --device controlboardwrapper2 --subdevice YarpOpenraveControlboard --robotIndex 0 --allManipulators
```

However, we must decide if the severity of #59 is blocking.

An alternative I do not like, due to forcing use of Python (extra dep, in addition to more complex debugging on segfaults), to is to map `teoSim` to:
```
python ~/repos/openrave-yarp-plugins/examples/openraveYarpPluginLoader-controlboard-allManipulators.py
```

@PeterBowman What would your opinion be, how bad do you think #59 (and even #60) is/are?

PeterBowman, 2017-12-26:
@jgvictores #59 is quite bad, but also easy to bypass (see my comments there). Similarly, #60 only occurs under specific circumstances, so we could count on finding a workaround later and just keep working with the current code and tools.

jgvictores, 2018-01-05:
Incoming `scripts/bash/teoSim` as part of https://github.com/roboticslab-uc3m/teo-configuration-files

jgvictores, 2018-01-08:
WIP at https://github.com/roboticslab-uc3m/teo-configuration-files/commit/5626005 ([teo-configuration-files](https://github.com/roboticslab-uc3m/teo-configuration-files) branch `openrave-yarp-plugins-27`).

jgvictores, 2018-01-08:
New working and installed `teoSim` merged into `develop` at https://github.com/roboticslab-uc3m/teo-configuration-files/commit/7fde99ba4f34e7ea1606aba9e7ecfef544ac3ff7.

@PeterBowman How should we proceed to make this transition minimally disruptive?

jgvictores, 2018-01-09:
No longer blocked. Only affected by #59 and #60 of this repo, which are also referenced from new teo-main issue https://github.com/roboticslab-uc3m/teo-main/issues/28

jgvictores, 2018-01-09:
Will close probably some day this week, after removing `teoSim` from this repository, in favor of the https://github.com/roboticslab-uc3m/teo-configuration-files script.

jgvictores, 2018-01-09:
Note to self: do not close until seeing things like `depthToOR` commented above.

jgvictores, 2018-01-11:
Blocks https://github.com/roboticslab-uc3m/teo-openrave-models/issues/10 and probably several others

jgvictores, 2018-01-12:
...`teoSim` deleted as of 08f5d130c356401a448b33bee38fbeff69a76a5d !!! :tada::tada::tada:

Will open new issues if anything missing from above.  Closing at last!

closed 2018-01-12 by jgvictores