roboticslab-uc3m/openrave-yarp-plugins #28: Error: undefined symbol: add_yarpplugins_plugins

rsantos88, 2017-05-15:
Trying to run the example openrave-YarpRobot.py in my personal fork... it breaks showing this:

```bash
teo@teo-oliver:~/repos/rsantos88/openrave-yarp-plugins/examples$ python openrave-YarpRobot.py 
[plugindatabase.h:577 Create] Failed to create name fcl_, interface collisionchecker
[xmlreaders-core.cpp:2591 endElement] failed to create iksolver ik/ikfast0x10000048.Transform6D.x86_64.16_17_18_19_20_21.so
[xmlreaders-core.cpp:2591 endElement] failed to create iksolver ik/ikfast0x10000048.Transform6D.x86_64.22_23_24_25_26_27.so
python: symbol lookup error: /usr/local/lib/openrave0.9-plugins/libOpenraveYarpControlboard.so: undefined symbol: add_yarpplugins_plugins
```

jgvictores, 2017-05-15:
Could you try the `fix-28-linker-error` branch I just created?
(attempt of fix at 548374990a44864ba2a20b2d802f72c19aad81c8)

rsantos88, 2017-05-15:
Yes!! It's running perfect now.. thanks @jgvictores !

jgvictores, 2017-05-15:
Great! Please merge into `develop` if you can.

rsantos88, 2017-05-15:
yes, of course! :+1:

closed 2017-05-15 by rsantos88

RaulFdzbis, 2017-10-10:
I had this same error in a new fresh computer, the solution was again to do the same as in https://github.com/roboticslab-uc3m/openrave-yarp-plugins/commit/548374990a44864ba2a20b2d802f72c19aad81c8. 

Is not this solution merged with develop yet?

jgvictores, 2017-10-10:
It looks like it has been merged with `develop`, see: https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/develop/openraveplugins/OpenraveYarpControlboard/OpenraveYarpControlboard.cpp

Could you do the following? 
- sudo rm /usr/local/lib/openrave0.9-plugins/libOpenraveYarpControlboard
- git pull  # copy+paste output here
- git status # copy+paste output here
- make && sudo make install
- python openrave-YarpRobot.py # copy+paste output here