roboticslab-uc3m/openrave-yarp-plugins #35: Openrave Errors executing examples

rsantos88, 2017-09-12:
As some people know, I am trying to run correctly the openrave official examples.

Many examples are very useful for me with the propose to apply this cases in a future project with Teo.
But unfortunately I have been trying for a long time to execute correctly this demonstrations and it has been impossible. 
I've tried to install openrave and its dependences in a clean distribution of Ubuntu (I've tried Ubuntu 14.04 LTS and Ubuntu 16.04 LTS 64 bits) but the error persist with the same problem.
For example, openrave.py --example dualarmdemo_schunk 
(http://openrave.org/docs/0.8.2/openravepy/examples.dualarmdemo_schunk/) this is the result:

[image]

There are one guy who have this machine and it runs perfectly...
Computer: Asus Laptop, G501V
Processor: Intel® Core™ i7
Video Card: GeForce GTX 960M
RAM: 24 GB
OS: Ubuntu 16.04 (Xenial), 64 bits

[image]

I don't know what can be wrong...
If somebody could try to install Openrave and say to me his results, I'll be very grateful.
Here there are the scripts to install openrave and its dependences:
https://github.com/crigroup/openrave-installation

jgvictores, 2017-09-12:
The first warning is related to OSG, maybe we could focus first on that!
How about copying and pasting the result of the following?
`sudo updatedb && locate libosgDB.so`

PeterBowman, 2017-09-12:
Which version of OpenRAVE are you running on? If it's `latest_stable` (as in our installation guides, which, per Rosen Diankov, is [over 5 years old](https://github.com/rdiankov/openrave/issues/296#issuecomment-324598804)), I'd give a try to current `master`.

jgvictores, 2017-09-12:
@PeterBowman I've started https://github.com/roboticslab-uc3m/installation-guides/issues/21 related to that issue; it's actually something I've been thinking about for a while.

rsantos88, 2017-09-12:
@jgvictores the result of ` sudo updatedb && locate libosgDB.so` s:
```bash
/home/teo/git/OpenSceneGraph/build/lib/libosgDB.so
/home/teo/git/OpenSceneGraph/build/lib/libosgDB.so.131
/home/teo/git/OpenSceneGraph/build/lib/libosgDB.so.3.4.2
/usr/lib/x86_64-linux-gnu/libosgDB.so
/usr/lib/x86_64-linux-gnu/libosgDB.so.100
/usr/lib/x86_64-linux-gnu/libosgDB.so.3.2.1
/usr/local/lib64/libosgDB.so
/usr/local/lib64/libosgDB.so.131
/usr/local/lib64/libosgDB.so.3.4.2
```
@PeterBowman Yes, I've tried with the latest_stable branch and master and it's the same...

jgvictores, 2017-09-12:
Ok, let's see: `echo $LD_LIBRARY_PATH`

PeterBowman, 2017-09-12:
Note the version mismatch (`libosgDB.so.130` vs `libosgDB.so.131`).

rsantos88, 2017-09-12:
echo $LD_LIBRARY_PATH:
```bash
/home/teo/ros_catkin_ws/install_isolated/lib:/usr/local/lib:/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_
```

rsantos88, 2017-09-12:
> The first warning is related to OSG, maybe we could focus first on that!
How about copying and pasting the result of the following?
sudo updatedb && locate libosgDB.so

Ok, that problem is solved. The main problem is the "bad joint resolution 0.000000" message.
This is the bash error:
```bash
teo@teo-oliver:~/git/openrave/build$ openrave.py --example dualarmdemo_schunk
2017-09-12 14:44:48,671 openrave [WARN] [colladareader.cpp:2021 KinBody::LinkPtr OpenRAVE::ColladaReader::ExtractLink] bad joint resolution 0.000000
2017-09-12 14:44:48,671 openrave [WARN] [colladareader.cpp:2021 KinBody::LinkPtr OpenRAVE::ColladaReader::ExtractLink] bad joint resolution 0.000000
2017-09-12 14:44:48,671 openrave [WARN] [colladareader.cpp:2021 KinBody::LinkPtr OpenRAVE::ColladaReader::ExtractLink] bad joint resolution 0.000000
2017-09-12 14:44:48,671 openrave [WARN] [colladareader.cpp:2021 KinBody::LinkPtr OpenRAVE::ColladaReader::ExtractLink] bad joint resolution 0.000000
2017-09-12 14:44:48,671 openrave [WARN] [colladareader.cpp:2021 KinBody::LinkPtr OpenRAVE::ColladaReader::ExtractLink] bad joint resolution 0.000000
2017-09-12 14:44:48,671 openrave [WARN] [colladareader.cpp:2021 KinBody::LinkPtr OpenRAVE::ColladaReader::ExtractLink] bad joint resolution 0.000000
2017-09-12 14:44:48,671 openrave [WARN] [colladareader.cpp:2021 KinBody::LinkPtr OpenRAVE::ColladaReader::ExtractLink] bad joint resolution 0.000000
2017-09-12 14:44:48,775 openrave [WARN] [colladareader.cpp:2021 KinBody::LinkPtr OpenRAVE::ColladaReader::ExtractLink] bad joint resolution 0.000000
2017-09-12 14:44:48,776 openrave [WARN] [colladareader.cpp:2021 KinBody::LinkPtr OpenRAVE::ColladaReader::ExtractLink] bad joint resolution 0.000000
2017-09-12 14:44:48,776 openrave [WARN] [colladareader.cpp:2021 KinBody::LinkPtr OpenRAVE::ColladaReader::ExtractLink] bad joint resolution 0.000000
2017-09-12 14:44:48,776 openrave [WARN] [colladareader.cpp:2021 KinBody::LinkPtr OpenRAVE::ColladaReader::ExtractLink] bad joint resolution 0.000000
2017-09-12 14:44:48,776 openrave [WARN] [colladareader.cpp:2021 KinBody::LinkPtr OpenRAVE::ColladaReader::ExtractLink] bad joint resolution 0.000000
2017-09-12 14:44:48,776 openrave [WARN] [colladareader.cpp:2021 KinBody::LinkPtr OpenRAVE::ColladaReader::ExtractLink] bad joint resolution 0.000000
2017-09-12 14:44:48,776 openrave [WARN] [colladareader.cpp:2021 KinBody::LinkPtr OpenRAVE::ColladaReader::ExtractLink] bad joint resolution 0.000000
openravepy.databases.inversekinematics: GetDefaultIndices, found 3-intersection centered on index 1
2017-09-12 14:44:51,752 openrave [ERROR] [ikfastsolver.cpp:256 Init] inverse kinematics hashes do not match for manip schunk-lwa3-dual:leftarm. IK will not work! e494c1cf5a076877522b8e32ef2817a3!=f1b4ece80cdeeec66467d9998ec73679
openravepy.databases.inversekinematics: GetDefaultIndices, found 3-intersection centered on index 8
2017-09-12 14:44:51,754 openrave [ERROR] [ikfastsolver.cpp:256 Init] inverse kinematics hashes do not match for manip schunk-lwa3-dual:rightarm. IK will not work! e494c1cf5a076877522b8e32ef2817a3!=f1b4ece80cdeeec66467d9998ec73679
Moving to Grasping position for object: <kinbody: - Object1 (bb33c57b05a8c4fae78731116b18548a)>
openravepy: RobotStateSaver, use body.CreateRobotStateSaver instead of RobotStateSaver
[OpenraveYarpPaintSquares.cpp:253 DestroyPlugin] destroying plugin
[OpenraveYarpControlboard.cpp:227 DestroyPlugin] destroying plugin
[OpenraveYarpCatchObject.cpp:150 DestroyPlugin] destroying plugin
[OpenraveWorldRpcResponder.cpp:664 DestroyPlugin] destroying plugin
Traceback (most recent call last):
  File "/usr/local/bin/openrave.py", line 149, in <module>
    example.run(args=args)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/openravepy_ext.py", line 64, in newfn
    return fn(*args,**kwargs)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/examples/dualarmdemo_schunk.py", line 184, in run
    OpenRAVEGlobalArguments.parseAndCreateThreadedUser(options,main,defaultviewer=True)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/misc.py", line 242, in parseAndCreateThreadedUser
    SetViewerUserThread(env,viewername,lambda: userfn(env,options))
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/misc.py", line 102, in SetViewerUserThread
    userfn()
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/misc.py", line 242, in <lambda>
    SetViewerUserThread(env,viewername,lambda: userfn(env,options))
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/examples/dualarmdemo_schunk.py", line 163, in main
    schunk.graspAndMoveObject(jointvalues,obj)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/examples/dualarmdemo_schunk.py", line 110, in graspAndMoveObject
    self.planDualPath(obj)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/examples/dualarmdemo_schunk.py", line 78, in planDualPath
    solutions = self.dualsolver.findMultiIKSolution(Tgrasps=[TLeftGrasp,TRightGrasp],filteroptions=IkFilterOptions.CheckEnvCollisions)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/misc.py", line 491, in findMultiIKSolution
    values=manip.FindIKSolutions(Tgrasps[i],filteroptions)
RuntimeError: tr1::bad_weak_ptr
```

rsantos88, 2017-09-12:
I am running the `master` version of openrave.

David-Estevez, 2017-09-13:
Same thing here (Linux Mint 17.3):

I think errors are related to Inverse Kinematics .

```
def@def-uc3m ~/git/openrave/build $ openrave.py --example dualarmdemo_schunk
[colladareader.cpp:2020 ExtractLink] bad joint resolution 0.000000
[colladareader.cpp:2020 ExtractLink] bad joint resolution 0.000000
[colladareader.cpp:2020 ExtractLink] bad joint resolution 0.000000
[colladareader.cpp:2020 ExtractLink] bad joint resolution 0.000000
[colladareader.cpp:2020 ExtractLink] bad joint resolution 0.000000
[colladareader.cpp:2020 ExtractLink] bad joint resolution 0.000000
[colladareader.cpp:2020 ExtractLink] bad joint resolution 0.000000
[colladareader.cpp:2020 ExtractLink] bad joint resolution 0.000000
[colladareader.cpp:2020 ExtractLink] bad joint resolution 0.000000
[colladareader.cpp:2020 ExtractLink] bad joint resolution 0.000000
[colladareader.cpp:2020 ExtractLink] bad joint resolution 0.000000
[colladareader.cpp:2020 ExtractLink] bad joint resolution 0.000000
[colladareader.cpp:2020 ExtractLink] bad joint resolution 0.000000
[colladareader.cpp:2020 ExtractLink] bad joint resolution 0.000000
openravepy.databases.inversekinematics: GetDefaultIndices, found 3-intersection centered on index 1
[ikfastsolver.cpp:254 Init] inverse kinematics hashes do not match for manip schunk-lwa3-dual:leftarm. IK will not work! e494c1cf5a076877522b8e32ef2817a3!=f1b4ece80cdeeec66467d9998ec73679
openravepy.databases.inversekinematics: GetDefaultIndices, found 3-intersection centered on index 8
[ikfastsolver.cpp:254 Init] inverse kinematics hashes do not match for manip schunk-lwa3-dual:rightarm. IK will not work! e494c1cf5a076877522b8e32ef2817a3!=f1b4ece80cdeeec66467d9998ec73679
Moving to Grasping position for object: <kinbody: - Object1 (bb33c57b05a8c4fae78731116b18548a)>
openravepy: RobotStateSaver, use body.CreateRobotStateSaver instead of RobotStateSaver
Traceback (most recent call last):
  File "/usr/local/bin/openrave.py", line 149, in <module>
    example.run(args=args)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/openravepy_ext.py", line 64, in newfn
    return fn(*args,**kwargs)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/examples/dualarmdemo_schunk.py", line 184, in run
    OpenRAVEGlobalArguments.parseAndCreateThreadedUser(options,main,defaultviewer=True)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/misc.py", line 240, in parseAndCreateThreadedUser
    SetViewerUserThread(env,viewername,lambda: userfn(env,options))
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/misc.py", line 100, in SetViewerUserThread
    userfn()
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/misc.py", line 240, in <lambda>
    SetViewerUserThread(env,viewername,lambda: userfn(env,options))
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/examples/dualarmdemo_schunk.py", line 163, in main
    schunk.graspAndMoveObject(jointvalues,obj)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/examples/dualarmdemo_schunk.py", line 110, in graspAndMoveObject
    self.planDualPath(obj)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/examples/dualarmdemo_schunk.py", line 78, in planDualPath
    solutions = self.dualsolver.findMultiIKSolution(Tgrasps=[TLeftGrasp,TRightGrasp],filteroptions=IkFilterOptions.CheckEnvCollisions)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/misc.py", line 489, in findMultiIKSolution
    values=manip.FindIKSolutions(Tgrasps[i],filteroptions)
RuntimeError: tr1::bad_weak_ptr
Segmentation fault
```

rsantos88, 2017-09-14:
> The first warning is related to OSG, maybe we could focus first on that!
How about copying and pasting the result of the following?
sudo updatedb && locate libosgDB.so

On request of @jgvictores , this error have been solved uninstalling and reinstalling all the dependences using this [scripts](https://github.com/crigroup/openrave-installation).

jgvictores, 2017-09-14:
Looking at:
> openravepy.databases.inversekinematics: GetDefaultIndices, found 3-intersection centered on index 1
2017-09-12 14:44:51,752 openrave [ERROR] [ikfastsolver.cpp:256 Init] inverse kinematics hashes do not match for manip schunk-lwa3-dual:leftarm. IK will not work! e494c1cf5a076877522b8e32ef2817a3!=f1b4ece80cdeeec66467d9998ec73679
openravepy.databases.inversekinematics: GetDefaultIndices, found 3-intersection centered on index 8
2017-09-12 14:44:51,754 openrave [ERROR] [ikfastsolver.cpp:256 Init] inverse kinematics hashes do not match for manip schunk-lwa3-dual:rightarm. IK will not work! e494c1cf5a076877522b8e32ef2817a3!=f1b4ece80cdeeec66467d9998ec73679

I'd try to focus on this IK hash mismatch. Maybe you could erase any existing schunk-lwa3-dual:leftarm and schunk-lwa3-dual:rightarm ikfast solver, and/or try to force the generation of a new one?

((note to self: focus on `openravepy: RobotStateSaver, use body.CreateRobotStateSaver instead of RobotStateSaver` after the hash issue is solved))

rsantos88, 2017-09-14:
After remove all the content of `.openrave` directory, the error persist:
```bash
openrave.py --example dualarmdemo_schunk
2017-09-14 12:38:42,636 openrave [WARN] [colladareader.cpp:2020 KinBody::LinkPtr OpenRAVE::ColladaReader::ExtractLink] bad joint resolution 0.000000
2017-09-14 12:38:42,636 openrave [WARN] [colladareader.cpp:2020 KinBody::LinkPtr OpenRAVE::ColladaReader::ExtractLink] bad joint resolution 0.000000
2017-09-14 12:38:42,636 openrave [WARN] [colladareader.cpp:2020 KinBody::LinkPtr OpenRAVE::ColladaReader::ExtractLink] bad joint resolution 0.000000
2017-09-14 12:38:42,637 openrave [WARN] [colladareader.cpp:2020 KinBody::LinkPtr OpenRAVE::ColladaReader::ExtractLink] bad joint resolution 0.000000
2017-09-14 12:38:42,637 openrave [WARN] [colladareader.cpp:2020 KinBody::LinkPtr OpenRAVE::ColladaReader::ExtractLink] bad joint resolution 0.000000
2017-09-14 12:38:42,637 openrave [WARN] [colladareader.cpp:2020 KinBody::LinkPtr OpenRAVE::ColladaReader::ExtractLink] bad joint resolution 0.000000
2017-09-14 12:38:42,637 openrave [WARN] [colladareader.cpp:2020 KinBody::LinkPtr OpenRAVE::ColladaReader::ExtractLink] bad joint resolution 0.000000
2017-09-14 12:38:42,742 openrave [WARN] [colladareader.cpp:2020 KinBody::LinkPtr OpenRAVE::ColladaReader::ExtractLink] bad joint resolution 0.000000
2017-09-14 12:38:42,742 openrave [WARN] [colladareader.cpp:2020 KinBody::LinkPtr OpenRAVE::ColladaReader::ExtractLink] bad joint resolution 0.000000
2017-09-14 12:38:42,742 openrave [WARN] [colladareader.cpp:2020 KinBody::LinkPtr OpenRAVE::ColladaReader::ExtractLink] bad joint resolution 0.000000
2017-09-14 12:38:42,742 openrave [WARN] [colladareader.cpp:2020 KinBody::LinkPtr OpenRAVE::ColladaReader::ExtractLink] bad joint resolution 0.000000
2017-09-14 12:38:42,742 openrave [WARN] [colladareader.cpp:2020 KinBody::LinkPtr OpenRAVE::ColladaReader::ExtractLink] bad joint resolution 0.000000
2017-09-14 12:38:42,742 openrave [WARN] [colladareader.cpp:2020 KinBody::LinkPtr OpenRAVE::ColladaReader::ExtractLink] bad joint resolution 0.000000
2017-09-14 12:38:42,742 openrave [WARN] [colladareader.cpp:2020 KinBody::LinkPtr OpenRAVE::ColladaReader::ExtractLink] bad joint resolution 0.000000
openravepy.databases.inversekinematics: GetDefaultIndices, found 3-intersection centered on index 1
2017-09-14 12:38:45,707 openrave [ERROR] [ikfastsolver.cpp:254 Init] inverse kinematics hashes do not match for manip schunk-lwa3-dual:leftarm. IK will not work! e494c1cf5a076877522b8e32ef2817a3!=f1b4ece80cdeeec66467d9998ec73679
openravepy.databases.inversekinematics: GetDefaultIndices, found 3-intersection centered on index 8
2017-09-14 12:38:45,715 openrave [ERROR] [ikfastsolver.cpp:254 Init] inverse kinematics hashes do not match for manip schunk-lwa3-dual:rightarm. IK will not work! e494c1cf5a076877522b8e32ef2817a3!=f1b4ece80cdeeec66467d9998ec73679
Moving to Grasping position for object: <kinbody: - Object1 (bb33c57b05a8c4fae78731116b18548a)>
openravepy: RobotStateSaver, use body.CreateRobotStateSaver instead of RobotStateSaver
[OpenraveYarpPaintSquares.cpp:253 DestroyPlugin] destroying plugin
[OpenraveYarpControlboard.cpp:227 DestroyPlugin] destroying plugin
[OpenraveYarpCatchObject.cpp:150 DestroyPlugin] destroying plugin
[OpenraveWorldRpcResponder.cpp:664 DestroyPlugin] destroying plugin
Traceback (most recent call last):
  File "/usr/local/bin/openrave.py", line 149, in <module>
    example.run(args=args)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/openravepy_ext.py", line 64, in newfn
    return fn(*args,**kwargs)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/examples/dualarmdemo_schunk.py", line 184, in run
    OpenRAVEGlobalArguments.parseAndCreateThreadedUser(options,main,defaultviewer=True)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/misc.py", line 240, in parseAndCreateThreadedUser
    SetViewerUserThread(env,viewername,lambda: userfn(env,options))
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/misc.py", line 100, in SetViewerUserThread
    userfn()
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/misc.py", line 240, in <lambda>
    SetViewerUserThread(env,viewername,lambda: userfn(env,options))
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/examples/dualarmdemo_schunk.py", line 163, in main
    schunk.graspAndMoveObject(jointvalues,obj)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/examples/dualarmdemo_schunk.py", line 110, in graspAndMoveObject
    self.planDualPath(obj)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/examples/dualarmdemo_schunk.py", line 78, in planDualPath
    solutions = self.dualsolver.findMultiIKSolution(Tgrasps=[TLeftGrasp,TRightGrasp],filteroptions=IkFilterOptions.CheckEnvCollisions)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/misc.py", line 489, in findMultiIKSolution
    values=manip.FindIKSolutions(Tgrasps[i],filteroptions)
RuntimeError: tr1::bad_weak_ptr
```

jgvictores, 2017-09-14:
Given `inverse kinematics hashes do not match for manip schunk-lwa3-dual:leftarm`, I still have the feeling there might be some pre-computed ikfast or something, perhaps in the repository and not in `.openrave`.

David-Estevez, 2017-09-15:
As @rsantos88 and I observed the other day in the lab, that error seems to appear as the simulator is loading the robot models incorrectly (and therefore it is computing IK for a garbage robot).

We first thought that the robot files might be corrupt, but opening them directly (outside of the simulation, from command line) loads the robot correctly. Loading them from the GUI, on the other hand, also loads a corrupt / garbage robot. This makes me think it might be a problem with collada dom parser.

rsantos88, 2017-09-15:
Finally and fortunately, the problem has been solved.   :tada:  :tada:  :tada:

As @David-Estevez said, I try to open the model directly with command line and the model is loaded perfectly and without any error. Not the same, when we load the model with the GUI. The result is this [bug](https://github.com/rdiankov/openrave/issues/435). And you can see that this problem is produced by the viewer. Why? Maybe the problem is not the type of viewer that you run... is the order in which you have to load the model and the viewer...
You can do this example. If you try to put this commands:

>  python -c "import openravepy as orpy; env = orpy.Environment(); env.Load('data/lab1.env.xml'); env.SetDefaultViewer(); raw_input('Press enter to exit...')"
python -c "import openravepy as orpy; env = orpy.Environment(); env.SetDefaultViewer(); env.Load('data/lab1.env.xml'); raw_input('Press enter to exit...')"

The second command should fail.
The workaround for this bug it to always load the environment (env.Load) before starting the viewer.
For this reason, I've modified the python example, changing the viewer ([changes](https://github.com/rsantos88/openrave/commit/4fa0a7aef3cba71d148232f981ef3049a148e19b)) and configuring the viewer after loading the model. 
It works great now!! But I don't know the real reason for this problem...

David-Estevez, 2017-09-18:
In my PC both commands seem to work. But this time the GUI viewer excuted was not the classical one (that I think it's QTCoin), but the Qtosg (Qt OpenSceneGraph) one:

[image]
 
I don't know if that might be related to the issue or not, but I'm commenting it here just in case it is useful for somebody.

rsantos88, 2017-09-18:
@David-Estevez  sorry, I've corrected the path of the model. The problem is with the `.dae` models, not with the `.xml`.
Try it now.

David-Estevez, 2017-09-18:
Ok, with `.dae` models it does fail. I have no clues about why.

jgvictores, 2017-10-31:
Can this issue be closed (or something more specific moved directly to https://github.com/rdiankov/openrave)?

rsantos88, 2017-11-14:
@jgvictores  yes, I'll close this issue. Thanks

closed 2017-11-14 by rsantos88

jgvictores, 2020-01-25:
> Ok, with `.dae` models it does fail. I have no clues about why.

Tiny recap: as mentioned at https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/35#issuecomment-329764676, the issue was loading `.dae` files, as in https://github.com/rdiankov/openrave/issues/435. To sum up:
```
I think you have run into this bug/issue:
https://github.com/rdiankov/openrave/issues/435

To make sure, try these two commands:

   1. python -c "import openravepy as orpy; env = orpy.Environment();
   env.Load('data/lab1.env.xml'); env.SetDefaultViewer(); raw_input('Press
   enter to exit...')"
   2. python -c "import openravepy as orpy; env = orpy.Environment();
   env.SetDefaultViewer(); env.Load('data/lab1.env.xml'); raw_input('Press
   enter to exit...')"

*The second command should fail.

The workaround for this bug it to *always *load the environment (env.Load)
*before* starting the viewer.
```

And:
```bash
openrave robots/mitsubishi-pa10.zae # fails
openrave.py robots/mitsubishi-pa10.zae # works
```

Also note that https://github.com/rdiankov/openrave/issues/435 mentions it may be Ubuntu 16.04-specific.