roboticslab-uc3m/openrave-yarp-plugins #43: openrave-YarpRobotPaintSquares.py crashing after last comits

RaulFdzbis, 2017-11-30:
So when trying to execute openrave-YarpRobotPaintSquares.py I got the following output:

```bash
[plugindatabase.h:645 Create] Failed to create name fcl_, interface collisionchecker
[OpenraveYarpControlboard.cpp:111 Open] penv: 0x3229ef0
[OpenraveYarpControlboard.cpp:121 Open] Robots[0]: teoSim
[OpenraveYarpControlboard.cpp:129 Open] * Manipulators[0]: trunk
[OpenraveYarpControlboard.cpp:144 Open] * manipulatorPortName: /teoSim/trunk
[DEBUG]ControlBoardWrapper2: 'period' parameter missing, using default thread period = 20ms 
[DEBUG]opening controlBoardWrapper2 subdevice
[debug] DeviceDriverImpl.cpp:30 open(): config: (penv {240 158 34 3 0 0 0 0 192 165 34 3 0 0 0 0}) (robotIndex 0) (manipulatorIndex 0) (name "/teoSim/trunk") (device YarpOpenraveControlboard) (subdevice YarpOpenraveControlboard)
[debug] DeviceDriverImpl.cpp:182 open(): Get JointPtr for manipulatorIDs[0]: 0
[debug] DeviceDriverImpl.cpp:182 open(): Get JointPtr for manipulatorIDs[1]: 1
[debug] DeviceDriverImpl.cpp:198 open(): pcontrol: 0x3438380, IdealController
[info] DeviceDriverImpl.cpp:204 open(): Detected idealcontroller, switch to genericmulticontroller.
[debug] DeviceDriverImpl.cpp:217 open(): pcontrol: 0x546dac0, genericmulticontroller
[debug] DeviceDriverImpl.cpp:227 open(): Attach individual controller for manipulatorIDs[0]: 0
[debug] DeviceDriverImpl.cpp:227 open(): Attach individual controller for manipulatorIDs[1]: 1
[INFO]created device <YarpOpenraveControlboard>. See C++ class roboticslab::YarpOpenraveControlboard for documentation.
[info] IPositionImpl.cpp:8 getAxes(): Reporting 0 axes are present
[DEBUG]joints parameter is 2
[info] IPositionImpl.cpp:8 getAxes(): Reporting 0 axes are present
[info] IPositionImpl.cpp:8 getAxes(): Reporting 1 axes are present
[INFO]/teoSim/trunk : no ROS initialization required 
[INFO]/teoSim/trunk  initting YARP initialization 
yarp: Port /teoSim/trunk/rpc:i active at tcp://163.117.150.58:10002/
yarp: Port /teoSim/trunk/command:i active at tcp://163.117.150.58:10003/
yarp: Port /teoSim/trunk/state:o active at tcp://163.117.150.58:10004/
yarp: Port /teoSim/trunk/stateExt:o active at tcp://163.117.150.58:10005/
[INFO]created wrapper <controlboardwrapper2>. See C++ class yarp::dev::ControlBoardWrapper2 for documentation.
[OpenraveYarpControlboard.cpp:129 Open] * Manipulators[1]: head
[OpenraveYarpControlboard.cpp:144 Open] * manipulatorPortName: /teoSim/head
[DEBUG]ControlBoardWrapper2: 'period' parameter missing, using default thread period = 20ms 
[DEBUG]opening controlBoardWrapper2 subdevice
[debug] DeviceDriverImpl.cpp:30 open(): config: (penv {240 158 34 3 0 0 0 0 192 165 34 3 0 0 0 0}) (robotIndex 0) (manipulatorIndex 1) (name "/teoSim/head") (device YarpOpenraveControlboard) (subdevice YarpOpenraveControlboard)
[debug] DeviceDriverImpl.cpp:182 open(): Get JointPtr for manipulatorIDs[0]: 2
[debug] DeviceDriverImpl.cpp:182 open(): Get JointPtr for manipulatorIDs[1]: 3
[debug] DeviceDriverImpl.cpp:198 open(): pcontrol: 0x546dac0, genericmulticontroller
[info] DeviceDriverImpl.cpp:210 open(): Detected genericmulticontroller, which will be used.
[debug] DeviceDriverImpl.cpp:217 open(): pcontrol: 0x546dac0, genericmulticontroller
[debug] DeviceDriverImpl.cpp:227 open(): Attach individual controller for manipulatorIDs[0]: 2
[debug] DeviceDriverImpl.cpp:227 open(): Attach individual controller for manipulatorIDs[1]: 3
[INFO]created device <YarpOpenraveControlboard>. See C++ class roboticslab::YarpOpenraveControlboard for documentation.
[info] IPositionImpl.cpp:8 getAxes(): Reporting 0 axes are present
[DEBUG]joints parameter is 2
[info] IPositionImpl.cpp:8 getAxes(): Reporting 0 axes are present
[info] IPositionImpl.cpp:8 getAxes(): Reporting -1579889312 axes are present
[INFO]/teoSim/head : no ROS initialization required 
[INFO]/teoSim/head  initting YARP initialization 
yarp: Port /teoSim/head/rpc:i active at tcp://163.117.150.58:10006/
yarp: Port /teoSim/head/command:i active at tcp://163.117.150.58:10007/
yarp: Port /teoSim/head/state:o active at tcp://163.117.150.58:10008/
yarp: Port /teoSim/head/stateExt:o active at tcp://163.117.150.58:10009/
[INFO]created wrapper <controlboardwrapper2>. See C++ class yarp::dev::ControlBoardWrapper2 for documentation.
[OpenraveYarpControlboard.cpp:129 Open] * Manipulators[2]: rightArm
[OpenraveYarpControlboard.cpp:144 Open] * manipulatorPortName: /teoSim/rightArm
[DEBUG]ControlBoardWrapper2: 'period' parameter missing, using default thread period = 20ms 
[DEBUG]opening controlBoardWrapper2 subdevice
[debug] DeviceDriverImpl.cpp:30 open(): config: (penv {240 158 34 3 0 0 0 0 192 165 34 3 0 0 0 0}) (robotIndex 0) (manipulatorIndex 2) (name "/teoSim/rightArm") (device YarpOpenraveControlboard) (subdevice YarpOpenraveControlboard)
[debug] DeviceDriverImpl.cpp:182 open(): Get JointPtr for manipulatorIDs[0]: 4
[debug] DeviceDriverImpl.cpp:182 open(): Get JointPtr for manipulatorIDs[1]: 5
[debug] DeviceDriverImpl.cpp:182 open(): Get JointPtr for manipulatorIDs[2]: 6
[debug] DeviceDriverImpl.cpp:182 open(): Get JointPtr for manipulatorIDs[3]: 7
[debug] DeviceDriverImpl.cpp:182 open(): Get JointPtr for manipulatorIDs[4]: 8
[debug] DeviceDriverImpl.cpp:182 open(): Get JointPtr for manipulatorIDs[5]: 9
[debug] DeviceDriverImpl.cpp:198 open(): pcontrol: 0x546dac0, genericmulticontroller
[info] DeviceDriverImpl.cpp:210 open(): Detected genericmulticontroller, which will be used.
[debug] DeviceDriverImpl.cpp:217 open(): pcontrol: 0x546dac0, genericmulticontroller
[debug] DeviceDriverImpl.cpp:227 open(): Attach individual controller for manipulatorIDs[0]: 4
[debug] DeviceDriverImpl.cpp:227 open(): Attach individual controller for manipulatorIDs[1]: 5
[... 33 lines cut ...]
[OpenraveYarpPaintSquares.cpp:255 DestroyPlugin] destroying plugin
[OpenraveWorldRpcResponder.cpp:664 DestroyPlugin] destroying plugin
[OpenraveYarpForceEstimator.cpp:183 DestroyPlugin] destroying plugin
[OpenraveYarpForce.cpp:421 DestroyPlugin] destroying plugin
[qtcoinrave.cpp:80 DestroyPlugin] SoQt releasing all memory
Coin debug: socontexthandler_cleanup(): 231 context-bound resources not free'd before exit.
[OpenraveYarpControlboard.cpp:199 DestroyPlugin] destroying plugin
[OpenraveYarpPaintSquares.cpp:255 DestroyPlugin] destroying plugin
Traceback (most recent call last):
  File "openrave-YarpRobotPaintSquares.py", line 20, in <module>
    print OpenraveYarpControlboard.SendCommand('open')
openravepy._openravepy_.openravepy_ext.openrave_exception: openrave (Failed): controller already attached to dof 0
[YarpOpenraveControlboard] close()
[YarpOpenraveControlboard] close()
[YarpOpenraveControlboard] close()
[OpenraveYarpControlboard.cpp:199 DestroyPlugin] destroying plugin
```

Few notes:

1. Is this an expected behaviour:

>> [info] IPositionImpl.cpp:8 getAxes(): Reporting -1579889312 axes are present

2. I have also tried *openrave-YarpForceEstimator.py* and it works.

AlvaroMartinezR, 2017-11-30:
xddddd

jgvictores, 2017-11-30:
> [info] IPositionImpl.cpp:8 getAxes(): Reporting -1579889312 axes are present

Should be fixed at 6630ae95d82bc2c1768f986039d22b270d6655b3. Was a visualization error, shouldn't affect behavior.

When breaking, which model (repository+branch) are you using?

RaulFdzbis, 2017-11-30:
>>When breaking, which model (repository+branch) are you using?

openrave-yarp-plugins at develop.

jgvictores, 2017-11-30:
> openrave-yarp-plugins at develop

I refer to specific openrave model (environment, robot) file (shouldn't be models in openrave-yarp-plugins).

jgvictores, 2017-11-30:
This is, which branch of `xgnitive`? (please make sure latest changes are pushed)

RaulFdzbis, 2017-11-30:
>I refer to specific openrave model (environment, robot) file (shouldn't be models in openrave-yarp-plugins).

Ok, that was the problem a quick git pull of teo-openrave-models solved the problem. Thank you @jgvictores. This can be closed :D

closed 2017-11-30 by RaulFdzbis

jgvictores, 2017-11-30:
@RaulFdzbis Ok, I also commited d433cc5d05455fb12c20ef226f72029f04be73af which is better, but you'll have to install xgnitive to work with the latest model.