roboticslab-uc3m/openrave-yarp-plugins #44: Error while loading openraveYarpForceEstimator plugin

RaulFdzbis, 2017-11-30:
So when trying to load the openraveYarpForceEstimator. The code i am using to load the plugin is from xgnitive, can be checked [here](https://github.com/roboticslab-uc3m/xgnitive/blob/d5daac6d42416e70f9de4da2c6e53a3864f78a89/programs/drl/envs/teo_iron_discrete_yarp_env.py#L61-L73).

The output in the console is the following:

```bash
/home/raul/anaconda2/envs/rllab3/lib/python3.5/site-packages/theano/tensor/signal/downsample.py:6: UserWarning: downsample module has been moved to the theano.tensor.signal.pool module.
  "downsample module has been moved to the theano.tensor.signal.pool module.")
python /home/raul/repos/rllab/scripts/run_experiment_lite.py  --variant_data 'gAN9cQBYCAAAAGV4cF9uYW1lcQFYIwAAAGV4cGVyaW1lbnRfMjAxN18xMV8zMF8xOF8wMl81NF8wMDAxcQJzLg=='  --args_data 'gAJjY2xvdWRwaWNrbGUuY2xvdWRwaWNrbGUKX2ZpbGxfZnVuY3Rpb24KcQAoY2Nsb3VkcGlja2xlLmNsb3VkcGlja2xlCl9tYWtlX3NrZWxfZnVuYwpxAWNjbG91ZHBpY2tsZS5jbG91ZHBpY2tsZQpfYnVpbHRpbl90eXBlCnECWAgAAABDb2RlVHlwZXEDhXEEUnEFKEsASwBLBUsPS0djX2NvZGVjcwplbmNvZGUKcQZYegAAAHQAAHQBAMKDAADCgwEAfQEAdAIAZAEAfAEAagMAwoMAAX0CAHQEAGQBAHwBAGoDAMKDAAF9AwB0BQBkAgB8AQBkAwB8AgBkBAB8AwBkBQBkBgBkBwBkCABkCQBkCgBkCwBkDADCgwAHfQQAfAQAagYAwoMAAAFkAABTcQdYBgAAAGxhdGluMXEIhnEJUnEKKE5YCAAAAGVudl9zcGVjcQtYAwAAAGVudnEMWAYAAABwb2xpY3lxDVgIAAAAYmFzZWxpbmVxDlgKAAAAYmF0Y2hfc2l6ZXEPTegDWA8AAABtYXhfcGF0aF9sZW5ndGhxEE2QAVgIAAAAZGlzY291bnRxEUc/764UeuFHrlgJAAAAc3RlcF9zaXplcRJHP4R64UeuFHt0cRMoWAkAAABub3JtYWxpemVxFFgWAAAAVGVvSXJvbkRpc2NyZXRlWWFycEVudnEVWBQAAABDYXRlZ29yaWNhbE1MUFBvbGljeXEWWAQAAABzcGVjcRdYFQAAAExpbmVhckZlYXR1cmVCYXNlbGluZXEYWAQAAABUUlBPcRlYBQAAAHRyYWlucRp0cRsoWAEAAABfcRxoDGgNaA5YBAAAAGFsZ29xHXRxHlgZAAAAdGVvX2lyb25fZGlzY3JldGVfeWFycC5weXEfWAgAAABydW5fdGFza3EgSwloBlgaAAAAAAEPAQYBDAISAQYBBgEGAQYBBgEGAQYBCQVxIWgIhnEiUnEjKSl0cSRScSVdcSZ9cSeHcShScSl9cSooaBVjZW52cy50ZW9faXJvbl9kaXNjcmV0ZV95YXJwX2VudgpUZW9Jcm9uRGlzY3JldGVZYXJwRW52CnEraBljcmxsYWIuYWxnb3MudHJwbwpUUlBPCnEsaBhjcmxsYWIuYmFzZWxpbmVzLmxpbmVhcl9mZWF0dXJlX2Jhc2VsaW5lCkxpbmVhckZlYXR1cmVCYXNlbGluZQpxLWgWY3JsbGFiLnBvbGljaWVzLmNhdGVnb3JpY2FsX21scF9wb2xpY3kKQ2F0ZWdvcmljYWxNTFBQb2xpY3kKcS5oFGNybGxhYi5lbnZzLm5vcm1hbGl6ZWRfZW52Ck5vcm1hbGl6ZWRFbnYKcS91Tn1xMHRSLg=='  --use_cloudpickle 'True'  --snapshot_mode 'last'  --n_parallel '1'  --exp_name 'experiment_2017_11_30_18_02_54_0001'  --log_dir '/home/raul/repos/rllab/data/local/experiment/experiment_2017_11_30_18_02_54_0001'
/home/raul/anaconda2/envs/rllab3/lib/python3.5/site-packages/theano/tensor/signal/downsample.py:6: UserWarning: downsample module has been moved to the theano.tensor.signal.pool module.
  "downsample module has been moved to the theano.tensor.signal.pool module.")
[DEBUG]ControlBoardWrapper2: 'period' parameter missing, using default thread period = 20ms 
[DEBUG]opening controlBoardWrapper2 subdevice
[debug] DeviceDriverImpl.cpp:30 open(): config: (orPlugins (OpenraveWorldRpcResponder (module OpenraveWorldRpcResponder) (commands "open /drl/openraveWorldRpcResponder/rpc:s")) (OpenraveYarpForceEstimator (module OpenraveYarpForceEstimator) (commands "open /drl/openraveYarpForceEstimator/rpc:s"))) (env "/usr/local/share/xgnitive/contexts/models/teo_cgda_iros.env.xml") (robotIndex 0) (manipulatorIndex 2) (genRefSpeed 9999999) (name "/drl/rightArm") (device YarpOpenraveControlboard) (subdevice YarpOpenraveControlboard)
[debug] DeviceDriverImpl.cpp:42 open(): Found --env parameter.
[plugindatabase.h:645 Create] Failed to create name fcl_, interface collisionchecker
[success] DeviceDriverImpl.cpp:63 open(): Loaded environment '/usr/local/share/xgnitive/contexts/models/teo_cgda_iros.env.xml'.
[debug] DeviceDriverImpl.cpp:113 open(): Found --orPlugins parameter.
[debug] DeviceDriverImpl.cpp:121 open(): orPlugins: orPlugins (OpenraveWorldRpcResponder (module OpenraveWorldRpcResponder) (commands "open /drl/openraveWorldRpcResponder/rpc:s")) (OpenraveYarpForceEstimator (module OpenraveYarpForceEstimator) (commands "open /drl/openraveYarpForceEstimator/rpc:s"))
[debug] DeviceDriverImpl.cpp:130 open(): orPlugin[1]: OpenraveWorldRpcResponder (module OpenraveWorldRpcResponder) (commands "open /drl/openraveWorldRpcResponder/rpc:s")
[debug] DeviceDriverImpl.cpp:132 open(): * orPlugin[1]: plugin: OpenraveWorldRpcResponder
[debug] DeviceDriverImpl.cpp:134 open(): * orPlugin[1]: module: OpenraveWorldRpcResponder
[debug] DeviceDriverImpl.cpp:137 open(): * orPlugin[1]: commands: open /drl/openraveWorldRpcResponder/rpc:s
[OpenraveWorldRpcResponder.cpp:608 Open] portName: /drl/openraveWorldRpcResponder/rpc:s
[OpenraveWorldRpcResponder.cpp:617 Open] penv: 0x3bb29c0
OpenRaveWorldRpcResponder using robot 0 (teoSim) as main robot.
yarp: Port /drl/openraveWorldRpcResponder/rpc:s active at tcp://163.117.150.58:10040/
[success] DeviceDriverImpl.cpp:160 open(): Sent 'open /drl/openraveWorldRpcResponder/rpc:s' command.
[debug] DeviceDriverImpl.cpp:130 open(): orPlugin[2]: OpenraveYarpForceEstimator (module OpenraveYarpForceEstimator) (commands "open /drl/openraveYarpForceEstimator/rpc:s")
[debug] DeviceDriverImpl.cpp:132 open(): * orPlugin[2]: plugin: OpenraveYarpForceEstimator
[debug] DeviceDriverImpl.cpp:134 open(): * orPlugin[2]: module: OpenraveYarpForceEstimator
[debug] DeviceDriverImpl.cpp:137 open(): * orPlugin[2]: commands: open /drl/openraveYarpForceEstimator/rpc:s
[OpenraveYarpForceEstimator.cpp:138 Open] portName: /drl/openraveYarpForceEstimator/rpc:s
yarp: Port /drl/openraveYarpForceEstimator/rpc:s active at tcp://163.117.150.58:10041/
[OpenraveYarpForceEstimator.cpp:147 Open] penv: 0x3bb29c0
Robot 0: teoSim
[success] DeviceDriverImpl.cpp:160 open(): Sent 'open /drl/openraveYarpForceEstimator/rpc:s' command.
[debug] DeviceDriverImpl.cpp:182 open(): Get JointPtr for manipulatorIDs[0]: 4
[debug] DeviceDriverImpl.cpp:182 open(): Get JointPtr for manipulatorIDs[1]: 5
[debug] DeviceDriverImpl.cpp:182 open(): Get JointPtr for manipulatorIDs[2]: 6
[debug] DeviceDriverImpl.cpp:182 open(): Get JointPtr for manipulatorIDs[3]: 7
[debug] DeviceDriverImpl.cpp:182 open(): Get JointPtr for manipulatorIDs[4]: 8
[debug] DeviceDriverImpl.cpp:182 open(): Get JointPtr for manipulatorIDs[5]: 9
[debug] DeviceDriverImpl.cpp:198 open(): pcontrol: 0x4627fe0, IdealController
[info] DeviceDriverImpl.cpp:204 open(): Detected idealcontroller, switch to genericmulticontroller.
[debug] DeviceDriverImpl.cpp:217 open(): pcontrol: 0x3bff6a0, genericmulticontroller
[debug] DeviceDriverImpl.cpp:227 open(): Attach individual controller for manipulatorIDs[0]: 4
[debug] DeviceDriverImpl.cpp:227 open(): Attach individual controller for manipulatorIDs[1]: 5
[debug] DeviceDriverImpl.cpp:227 open(): Attach individual controller for manipulatorIDs[2]: 6
[debug] DeviceDriverImpl.cpp:227 open(): Attach individual controller for manipulatorIDs[3]: 7
[debug] DeviceDriverImpl.cpp:227 open(): Attach individual controller for manipulatorIDs[4]: 8
[debug] DeviceDriverImpl.cpp:227 open(): Attach individual controller for manipulatorIDs[5]: 9
[INFO]created device <YarpOpenraveControlboard>. See C++ class roboticslab::YarpOpenraveControlboard for documentation.
[info] IPositionImpl.cpp:9 getAxes(): Reporting 6 axes are present
[DEBUG]joints parameter is 6
[info] IPositionImpl.cpp:9 getAxes(): Reporting 6 axes are present
[info] IPositionImpl.cpp:9 getAxes(): Reporting 6 axes are present
[INFO]/drl/rightArm : no ROS initialization required 
[INFO]/drl/rightArm  initting YARP initialization 
terminate called after throwing an instance of 'OpenRAVE::openrave_exception'
  what():  openrave (Assert): [/usr/include/boost/smart_ptr/shared_ptr.hpp:653] -> typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = OpenRAVE::KinBody; typename boost::detail::sp_member_access<T>::type = OpenRAVE::KinBody*], expr: px != 0
Aborted (core dumped)
```

jgvictores, 2017-12-01:
Given [these lines](https://github.com/roboticslab-uc3m/xgnitive/blob/d5daac6d42416e70f9de4da2c6e53a3864f78a89/programs/drl/envs/teo_iron_discrete_yarp_env.py#L45-L46) of the specific commit to which you refer, it's strange seeing the output `(device YarpOpenraveControlboard) (subdevice YarpOpenraveControlboard)`.

Will try to look into this sometime in the evening, looks interesting. @RaulFdzbis Maybe you could answer this before I get to work on this:
- Are you sure you are using that specific, unmodified, commit?

jgvictores, 2017-12-01:
My current working theory is that the model loaded is incorrect.

Launching simply [examples/openrave-YarpForceEstimator.py](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/b54adbd99ec0afa05d5ab12e0877af50c6501d8a/examples/openrave-YarpForceEstimator.py) leads to:
```
  File "openrave-YarpForceEstimator.py", line 23, in <module>
    print OpenraveYarpForceEstimator.SendCommand('open')
openravepy._openravepy_.openravepy_ext.openrave_exception: openrave (InvalidArguments): [virtual bool OpenRAVE::RobotBase::Grab(OpenRAVE::KinBodyPtr, OpenRAVE::KinBody::LinkPtr):1480] (!!pbody) failed robot teoSim invalid bod to grab
```

Looking at the [libraries/OpenravePlugins/OpenraveYarpForceEstimator/OpenraveYarpForceEstimator](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/b54adbd99ec0afa05d5ab12e0877af50c6501d8a/libraries/OpenravePlugins/OpenraveYarpForceEstimator/OpenraveYarpForceEstimator.cpp#L40-L41) code, we find the hard-coded objects: `#define OBJECT_1 "object"` and `#define OBJECT_2 "TableKinBody"`.

It looks like [the model you are loading](https://github.com/roboticslab-uc3m/xgnitive/blob/d5daac6d42416e70f9de4da2c6e53a3864f78a89/programs/drl/envs/teo_iron_discrete_yarp_env.py#L47) does not contain those objects. In fact, we have to go to the `textiles-xgnitive` branch of textiles, [this commit](https://github.com/roboticslab-uc3m/textiles/commit/5124c48ccdd9ecdebd5c3c636b448fb6be67ef9d) to find something that may work.

jgvictores, 2017-12-01:
PS: Tried and at least opens without breaking!

RaulFdzbis, 2017-12-01:
Yes I think that's the bug. The correct environment should be the same as loaded in [openrave-YarpForceEstimator](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/b54adbd99ec0afa05d5ab12e0877af50c6501d8a/examples/openrave-YarpForceEstimator.py#L14), however we are using the [paint environment](https://github.com/roboticslab-uc3m/xgnitive/blob/d5daac6d42416e70f9de4da2c6e53a3864f78a89/programs/drl/envs/teo_iron_discrete_yarp_env.py#L45-L46) as you said.

 Maybe we should change the name of this xml to avoid future confussions, since **teo_cgda_iros.env.xml** seems pretty generic.

jgvictores, 2017-12-01:
>  Maybe we should change the name of this xml to avoid future confussions, since **teo_cgda_iros.env.xml** seems pretty generic.

Yup! The PAINT environment should definitely contain the word **paint** in its name! :tada:

jgvictores, 2017-12-03:
**Note to self:** Do not close issue until check on Grab has been incorporated.

RaulFdzbis, 2017-12-04:
Still not working for me :( . Although now the error is different.

```bash
/home/raul/anaconda2/envs/rllab3/lib/python3.5/site-packages/theano/tensor/signal/downsample.py:6: UserWarning: downsample module has been moved to the theano.tensor.signal.pool module.
  "downsample module has been moved to the theano.tensor.signal.pool module.")
python /home/raul/repos/rllab/scripts/run_experiment_lite.py  --exp_name 'experiment_2017_12_04_12_49_10_0001'  --args_data 'gAJjY2xvdWRwaWNrbGUuY2xvdWRwaWNrbGUKX2ZpbGxfZnVuY3Rpb24KcQAoY2Nsb3VkcGlja2xlLmNsb3VkcGlja2xlCl9tYWtlX3NrZWxfZnVuYwpxAWNjbG91ZHBpY2tsZS5jbG91ZHBpY2tsZQpfYnVpbHRpbl90eXBlCnECWAgAAABDb2RlVHlwZXEDhXEEUnEFKEsASwBLBUsPS0djX2NvZGVjcwplbmNvZGUKcQZYegAAAHQAAHQBAMKDAADCgwEAfQEAdAIAZAEAfAEAagMAwoMAAX0CAHQEAGQBAHwBAGoDAMKDAAF9AwB0BQBkAgB8AQBkAwB8AgBkBAB8AwBkBQBkBgBkBwBkCABkCQBkCgBkCwBkDADCgwAHfQQAfAQAagYAwoMAAAFkAABTcQdYBgAAAGxhdGluMXEIhnEJUnEKKE5YCAAAAGVudl9zcGVjcQtYAwAAAGVudnEMWAYAAABwb2xpY3lxDVgIAAAAYmFzZWxpbmVxDlgKAAAAYmF0Y2hfc2l6ZXEPTegDWA8AAABtYXhfcGF0aF9sZW5ndGhxEE2QAVgIAAAAZGlzY291bnRxEUc/764UeuFHrlgJAAAAc3RlcF9zaXplcRJHP4R64UeuFHt0cRMoWAkAAABub3JtYWxpemVxFFgWAAAAVGVvSXJvbkRpc2NyZXRlWWFycEVudnEVWBQAAABDYXRlZ29yaWNhbE1MUFBvbGljeXEWWAQAAABzcGVjcRdYFQAAAExpbmVhckZlYXR1cmVCYXNlbGluZXEYWAQAAABUUlBPcRlYBQAAAHRyYWlucRp0cRsoWAEAAABfcRxoDGgNaA5YBAAAAGFsZ29xHXRxHlgZAAAAdGVvX2lyb25fZGlzY3JldGVfeWFycC5weXEfWAgAAABydW5fdGFza3EgSwloBlgaAAAAAAEPAQYBDAISAQYBBgEGAQYBBgEGAQYBCQVxIWgIhnEiUnEjKSl0cSRScSVdcSZ9cSeHcShScSl9cSooaBljcmxsYWIuYWxnb3MudHJwbwpUUlBPCnEraBVjZW52cy50ZW9faXJvbl9kaXNjcmV0ZV95YXJwX2VudgpUZW9Jcm9uRGlzY3JldGVZYXJwRW52CnEsaBRjcmxsYWIuZW52cy5ub3JtYWxpemVkX2VudgpOb3JtYWxpemVkRW52CnEtaBhjcmxsYWIuYmFzZWxpbmVzLmxpbmVhcl9mZWF0dXJlX2Jhc2VsaW5lCkxpbmVhckZlYXR1cmVCYXNlbGluZQpxLmgWY3JsbGFiLnBvbGljaWVzLmNhdGVnb3JpY2FsX21scF9wb2xpY3kKQ2F0ZWdvcmljYWxNTFBQb2xpY3kKcS91Tn1xMHRSLg=='  --log_dir '/home/raul/repos/rllab/data/local/experiment/experiment_2017_12_04_12_49_10_0001'  --use_cloudpickle 'True'  --variant_data 'gAN9cQBYCAAAAGV4cF9uYW1lcQFYIwAAAGV4cGVyaW1lbnRfMjAxN18xMl8wNF8xMl80OV8xMF8wMDAxcQJzLg=='  --snapshot_mode 'last'  --n_parallel '1'
/home/raul/anaconda2/envs/rllab3/lib/python3.5/site-packages/theano/tensor/signal/downsample.py:6: UserWarning: downsample module has been moved to the theano.tensor.signal.pool module.
  "downsample module has been moved to the theano.tensor.signal.pool module.")
[DEBUG]ControlBoardWrapper2: 'period' parameter missing, using default thread period = 20ms 
[DEBUG]opening controlBoardWrapper2 subdevice
[debug] DeviceDriverImpl.cpp:30 open(): config: (orPlugins (OpenraveWorldRpcResponder (module OpenraveWorldRpcResponder) (commands "open /drl/openraveWorldRpcResponder/rpc:s")) (OpenraveYarpForceEstimator (module OpenraveYarpForceEstimator) (commands "open /drl/openraveYarpForceEstimator/rpc:s"))) (env "/home/raul/repos/textiles/textiles/ironing/manipulation/ironingSim/ironingSim.env.xml") (robotIndex 0) (manipulatorIndex 2) (genRefSpeed 9999999) (name "/drl/rightArm") (device YarpOpenraveControlboard) (subdevice YarpOpenraveControlboard)
[debug] DeviceDriverImpl.cpp:42 open(): Found --env parameter.
[plugindatabase.h:645 Create] Failed to create name fcl_, interface collisionchecker
[xmlreaders-core.cpp:2599 endElement] failed to create iksolver ik/ikfast0x10000048.Transform6D.x86_64.16_17_18_19_20_21.so
[xmlreaders-core.cpp:2599 endElement] failed to create iksolver ik/ikfast0x10000048.Transform6D.x86_64.22_23_24_25_26_27.so
[plugindatabase.h:300 CreateInterface] Create Interface: openrave exception , plugin /home/raul/repos/forceSensor/lib/libForceSensor.so: openrave (InvalidPlugin): [OpenRAVE::InterfaceBasePtr OpenRAVECreateInterface(OpenRAVE::InterfaceType, const string&, const char*, const char*, OpenRAVE::EnvironmentBasePtr):68] bad environment hash: bd2a389e2dd4d161673c4bb2fd592bf7!=fa5187466a171890bd7c004214317fe3
[plugindatabase.h:645 Create] Failed to create name forcesensor, interface sensor
[xmlreaders-core.cpp:1805 InterfaceXMLReader] xml readers failed to create instance of type sensor:forcesensor
[plugindatabase.h:560 Create] interface sensor name "" needs to start with a valid character
[xmlreaders-core.cpp:1843 _CheckInterface] xml readers failed to create instance of type sensors
[plugindatabase.h:560 Create] interface sensor name "" needs to start with a valid character
[xmlreaders-core.cpp:1843 _CheckInterface] xml readers failed to create instance of type sensors
[environment-core.h:501 Load] load failed on file /home/raul/repos/textiles/textiles/ironing/manipulation/ironingSim/ironingSim.env.xml
[error] DeviceDriverImpl.cpp:60 open(): Could not load '/home/raul/repos/textiles/textiles/ironing/manipulation/ironingSim/ironingSim.env.xml' environment.
yarpdev: ***ERROR*** driver <YarpOpenraveControlboard> was found but could not open
[ERROR]opening controlBoardWrapper2 subdevice... FAILED
[ERROR]ControlBoardWrapper: error while opening subdevice
yarpdev: ***ERROR*** driver <controlboardwrapper2> was found but could not open
Mental robot device not available.

TypeError: __init__() should return None, not 'int'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/raul/repos/rllab/scripts/run_experiment_lite.py", line 137, in <module>
    run_experiment(sys.argv)
  File "/home/raul/repos/rllab/scripts/run_experiment_lite.py", line 121, in run_experiment
    method_call(variant_data)
  File "teo_iron_discrete_yarp.py", line 10, in run_task
[... cut ...]
Also appear when executing   [openrave-YarpForceEstimator.py](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/b54adbd99ec0afa05d5ab12e0877af50c6501d8a/examples/openrave-YarpForceEstimator.py), however in this case the environment is loaded.

I think the error comes from two of this messages:

>[error] DeviceDriverImpl.cpp:60 open(): Could not load '/home/raul/repos/textiles/textiles/ironing/manipulation/ironingSim/ironingSim.env.xml' environment.
>yarpdev: ***ERROR*** driver YarpOpenraveControlboard was found but could not open

jgvictores, 2017-12-04:
The following looks bad:
```
[plugindatabase.h:300 CreateInterface] Create Interface: openrave exception , plugin /home/raul/repos/forceSensor/lib/libForceSensor.so: openrave (InvalidPlugin): [OpenRAVE::InterfaceBasePtr OpenRAVECreateInterface(OpenRAVE::InterfaceType, const string&, const char*, const char*, OpenRAVE::EnvironmentBasePtr):68] bad environment hash: bd2a389e2dd4d161673c4bb2fd592bf7!=fa5187466a171890bd7c004214317fe3
[plugindatabase.h:645 Create] Failed to create name forcesensor, interface sensor
[xmlreaders-core.cpp:1805 InterfaceXMLReader] xml readers failed to create instance of type sensor:forcesensor
[plugindatabase.h:560 Create] interface sensor name "" needs to start with a valid character
[xmlreaders-core.cpp:1843 _CheckInterface] xml readers failed to create instance of type sensors
[plugindatabase.h:560 Create] interface sensor name "" needs to start with a valid character
[xmlreaders-core.cpp:1843 _CheckInterface] xml readers failed to create instance of type sensors
[environment-core.h:501 Load] load failed on file /home/raul/repos/textiles/textiles/ironing/manipulation/ironingSim/ironingSim.env.xml
[error] DeviceDriverImpl.cpp:60 open(): Could not load '/home/raul/repos/textiles/textiles/ironing/manipulation/ironingSim/ironingSim.env.xml' environment.
```

Conflicts may be come from the virtual environment and the shared library of the forceSensor plugn path. I'd start by removing the unused forceSensor: this should avoid some related or un-related errors. It is safe to do so because we are using the forceSensorEstimator (which in turn does not depend on the external potentially problematic plugin).

jgvictores, 2018-01-12:
Only re-usable idea here is to check the Grab return value, which is already part of #66. Closing!

closed 2018-01-12 by jgvictores