roboticslab-uc3m/openrave-yarp-plugins #47: Create generic OpenraveYarpPlugin or similar

jgvictores, 2017-12-03:
As a follow-up of #30, a generic OpenRAVE plugin to load any YARP plugin should be developed:
- Could be named
- Code base could be extracted from `OpenraveYarpControlboard` and eventually superseed it.
- An important design decision: `OpenraveYarpControlboard` goes through all robots, getting all manipulators, opens a `YarpControlboard` for each one. That in fact emulates `teoSim`. Would we want something similar? Perhaps a good approach is to do this mega-approach, then allow only parts to be specified. Will try this first.

jgvictores, 2017-12-03:
Updated description, and set as technically blocking #30 as it may superseed it.

jgvictores, 2017-12-03:
Note that may superseed #30 which is blocking:
- https://github.com/asrob-uc3m/robotDevastation-robots/issues/36 
- Several https://github.com/roboticslab-uc3m/xgnitive tasks.
WIP at `issue-47-OpenraveYarpPluginLoader`.

jgvictores, 2017-12-03:
Very important notes on OpenRAVE plugin `main()` and `open()` at https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/30#issuecomment-306237545 as well as other ideas.

jgvictores, 2017-12-06:
`OpenraveYarpPluginLoader` that loads specific YARP device developed and merged into develop at 478142c.

Possible improvements:
- Multiple device load.
- Use `open` for auto-open from command line (must check for race conditions such as existing robot, see above).
- `name`: Work on manipulatorIndex/sensorIndex and other alternatives.
- `name`: Work on suffix.

jgvictores, 2017-12-22:
Implemented most improvements at 28ce5e9, including:
- Multiple device load.
- `name`: Work on manipulatorIndex/sensorIndex and other alternatives.

...which renders `OpenraveYarpControlboard` quite obsolete.

Considered not required:
- `name`: Work on suffix.

Would like to study:
- Use `open` for auto-open from command line (must check for race conditions such as existing robot, see [these comments](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/30#issuecomment-306237545)). However, according to [this](https://github.com/rdiankov/openrave/blob/ff43549fb6db281c7bc9a85794b88c29d6522ab4/include/openrave/module.h#L43-L48), "Robots might not necessarily be set before this function call"...

jgvictores, 2017-12-22:
Must also update main README.

jgvictores, 2017-12-23:
Announcement!

We no longer require Python to load YARP plugins via `OpenraveYarpPluginLoader`, thanks to d5cfee2.

Here are some details:

- The direct CLI one-liner succesor of `teoSim` is:
```bash
openrave /usr/local/share/teo-openrave-models/contexts/openrave/teo/teo.robot.xml --module OpenraveYarpPluginLoader "open --device controlboardwrapper2 --subdevice YarpOpenraveControlboard --robotIndex 0 --allManipulators"
```
- We can even do funky commands like the following, where `open` acts as a delimiter:
```bash
openrave /usr/local/share/teo-openrave-models/contexts/openrave/teo/teo.robot.xml --module OpenraveYarpPluginLoader "open --device controlboardwrapper2 --subdevice YarpOpenraveControlboard --robotIndex 0 --manipulatorIndex 0 open --device controlboardwrapper2 --subdevice YarpOpenraveControlboard --robotIndex 0 --manipulatorIndex 2"
```

- Two pending issues:
  - Via Python, destructors are clearly called upon `^C`. Via CLI, for some reason,  they are not called upon `^C`. The hack would be to set up catching the signal in this `main()` of the function, but maybe this is overkill and we should dig into OpenRAVE guts.
  - The commented above "Robots might not necessarily be set before this function call" issue has not appeared, but could appear as the result of some race condition as documented.

@PeterBowman Okay if I perform the following steps?
1. Update documentation taking these functionalities into account.
1. Delete `OpenraveYarpControlboard` and related documentation/examples. XGNITIVE examples have already been updated, so there should be no current users. Perhaps the most conflictive change, as I'm not a big fan of leaving obsolete and most probably unused modules around.
1. Put a two new issues, each regarding one of the pending issues above.
1. Close this issue, start commenting on others such as #27.

jgvictores, 2017-12-23:
PS: Also tried via `SendCommand` instead of direct call to `Open`, same non-destructor result.

jgvictores, 2017-12-23:
PPS: Racec condition does occur via `openrave.py`.

PeterBowman, 2017-12-23:
@jgvictores 👍

jgvictores, 2017-12-26:
Finished at 73061b0. Now continued at #59, #60 and #27.

closed 2017-12-26 by jgvictores