roboticslab-uc3m/openrave-yarp-plugins #48: Segmentation fault: cannot render offscreen (cartesianServer and any module that uses RaveBot/OpenRAVE cameras)

jgvictores, 2017-12-05:
_From @iagson on February 23, 2015 9:36_

Hello jgvictores,

I was trying to start an instance of a cartesianServer, to see if all software was installed correctly and the kitchen  appears to initialise upside down. That isn't so bad, but when you try to input some extra commands like " cartesianServer --env asibot_kitchen_cameras.env.xml " the GUI doesn't start beacuse of an error (already tried by changing the .ini file)

It's also true that I didn't install the asibot repository from github, but from Subversion, maybe that's the problem.

These are the screen prints I took:
[image]
[image]

Thank you very much.

_Copied from original issue: roboticslab-uc3m/asibot-main#3_

jgvictores, 2017-12-05:
1) On your first screenshot I see a YARP matching protocol error (1 0 0 vs 0 0 0). You should be able to fix that by cleaning and recompiling (may be overkill, but I'd first do it on YARP and then on ASIBOT).
2) Check if the environment that breaks has cameras. There is a known issue regarding the off-screen rendering we use and certain graphic card drivers (especially in Linux).

BTW, don't mind copying and pasting the entire log, it will make things easier!

jgvictores, 2017-12-05:
_From @iagson on February 23, 2015 12:17_

After recompiling both YARP and ASIBOT the error persists.
BTW, I had to rename the asibot-main directory to asibot so to initialise cartesianServer.
This is the whole log after trying to access a camera:

iagson@iagson-VirtualBox:~$ cartesianServer --env asibot_kitchen_cameras.env.xml
||| clearing context
||| adding context [cartesianServer/conf]
||| configuring
||| policy set to ASIBOT_ROOT
||| ASIBOT_ROOT: /home/iagson/asibot
||| loading policy from /home/iagson/asibot/ASIBOT_ROOT.ini
yarp: cannot read from /home/iagson/asibot/ASIBOT_ROOT.ini
||| loading policy from /home/iagson/.yarp/ASIBOT_ROOT.ini
yarp: cannot read from /home/iagson/.yarp/ASIBOT_ROOT.ini
||| loading policy from /etc/yarp/policies/ASIBOT_ROOT.ini
yarp: cannot read from /etc/yarp/policies/ASIBOT_ROOT.ini
||| loading policy from /etc/ASIBOT_ROOT.ini
yarp: cannot read from /etc/ASIBOT_ROOT.ini
||| loading policy from /usr/local/etc/ASIBOT_ROOT.ini
yarp: cannot read from /usr/local/etc/ASIBOT_ROOT.ini
||| failed to load policy from /home/iagson/asibot/ASIBOT_ROOT.ini /home/iagson/.yarp/ASIBOT_ROOT.ini /etc/yarp/policies/ASIBOT_ROOT.ini /etc/ASIBOT_ROOT.ini /usr/local/etc/ASIBOT_ROOT.ini
||| default config file specified as cartesianServer.ini
||| checking [/home/iagson/cartesianServer.ini](pwd)
||| checking [/home/iagson/.config/yarp/robots/default](robot YARP_CONFIG_HOME)
||| checking [/home/iagson/.local/share/yarp/robots/default](robot YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/robots/default](robot YARP_CONFIG_DIRS)
||| checking [/usr/share/upstart/xdg/yarp/robots/default](robot YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/robots/default](robot YARP_CONFIG_DIRS)
||| checking [/usr/share/ubuntu/yarp/robots/default](robot YARP_DATA_DIRS)
||| checking [/usr/share/gnome/yarp/robots/default](robot YARP_DATA_DIRS)
||| checking [/usr/local/share/yarp/robots/default](robot YARP_DATA_DIRS)
||| checking [/usr/share/yarp/robots/default](robot YARP_DATA_DIRS)
||| checking [/usr/share/ubuntu/yarp/config/path.d](robot path.d YARP_DATA_DIRS)
||| checking [/usr/share/gnome/yarp/config/path.d](robot path.d YARP_DATA_DIRS)
||| checking [/usr/local/share/yarp/config/path.d](robot path.d YARP_DATA_DIRS)
||| checking [/usr/share/yarp/config/path.d](robot path.d YARP_DATA_DIRS)
||| checking [/home/iagson/asibot/app/cartesianServer/conf/cartesianServer.ini](deprecated-old-style-context)
||| found /home/iagson/asibot/app/cartesianServer/conf/cartesianServer.ini
Run "cartesianServer --help" for options

## cartesianServer checking for yarp network... [ok]

CartesianServer using controller: cartesianbot,  prefix: /ravebot.

## CartesianServer using movjLocal: /ravebot/movjCartesianServer, movjRemote: /ravebot.

CartesianBot using A0: 0.300000, A1: 0.400000, A2: 0.400000, A3: 0.300000.
CartesianBot using cmcMs: 30.000000, duration: 20.000000, maxAcc: 0.200000, maxVel: 7.500000.
CartesianBot using robotDevice: controlboard, robotSubDevice: ravebot, robotName: /ravebot.
CartesianBot using robotLocal: N/A, robotRemote: N/A.

## [INFO]ControlBoard subdevice is ravebot

RaveBot using numMotors: 5, env: asibot_kitchen_cameras.env.xml, externObj: none, extraRobot: none.
RaveBot using genInitPos: 0.000000, genJointTol: 0.250000, genMaxLimit: 360.000000, genMinLimit: -360.000000.
RaveBot using genRefSpeed: 7.500000, genEncRawExposed: 0.017453, genVelRawExposed: 0.017453.
RaveBot using jmcMs: 20.000000, jmcMsAcc: 1.000000, modePosVel: 0, physics: none, viewer: 1.
RaveBot not using individual initPoss, defaulting to genInitPos.
RaveBot not using individual jointTols, defaulting to genJointTol.
RaveBot not using individual maxLimits, defaulting to genMaxLimit.
RaveBot not using individual minLimits, defaulting to genMinLimit.
RaveBot not using individual refSpeeds, defaulting to genRefSpeed.
RaveBot not using individual encRawExposeds, defaulting to genEncRawExposed.

## RaveBot not using individual velRawExposeds, defaulting to genVelRawExposed.

[plugindatabase.h:577 Create] Failed to create name viewerrecorder, interface module
[environment-core.h:800 SetPhysicsEngine] setting ode physics engine
[... 39 lines cut ...]
received an unknown request after a VOCAB_GET: prot
ERROR, expecting protocol 1 0 0, but remotecontrolboard returned protocol version 0 0 0
WARNING: ignoring error but please update YARP or the remotecontrolboard implementation
[RaveBot] getAxes reporting 5 axes are present
[INFO]created device <remote_controlboard>. See C++ class yarp::dev::RemoteControlBoard for documentation.
[CartesianServer] success: Acquired robot interfaces.
yarp: Port /ravebot/cartesianServer/rpc:i active at tcp://10.0.2.15:10010
yarp: Port /ravebot/cartesianServer/command:i active at tcp://10.0.2.15:10011
Coin info in glglue_allow_newer_opengl(): 

Features of OpenGL version > 1.0 has been
disabled, due to the use of a remote display.

This is so because many common OpenGL drivers
have problems in this regard.

To force full OpenGL use, set the environment
variable COIN_FULL_INDIRECT_RENDERING=1 and re-run the application.

If you don't want this message displayed again,
set the environment variable COIN_DONT_INFORM_INDIRECT_RENDERING=1.

[qtcoinviewer.cpp:3417 _GetCameraImage] offscreen renderer failed (check video driver), disabling
[qtcoinviewer.cpp:3370 _GetCameraImage] cannot render offscreen
Segmentation fault (core dumped)

jgvictores, 2017-12-05:
The final segmentation fault is due to "offscreen renderer failed (check video driver)".
Here are related issues:
http://openrave-users-list.185357.n3.nabble.com/qtcoin-viewer-error-quot-offscreen-renderer-failed-quot-td4025805.html
http://openrave-users-list.185357.n3.nabble.com/Vision-sensor-in-c-td2706454.html#a2735645
This is an old issue related to OpenRAVE and deeper. At the asibot-main level, all you can do is avoid simulated environments with cameras.

jgvictores, 2017-12-05:
Throughout old emails, I've commented that offscreen rendering was achived via Bumblebee's `optirun` (only applicable to Optimus NVIDIA technology I guess).

jgvictores, 2017-12-05:
On current openrave `master`, this currently manifests itself as:
```
Coin warning in glxglue_context_create_software(): Couldn't create GLX context.
[qtcoinviewer.cpp:3448 _GetCameraImage] offscreen renderer failed (check video driver), disabling
[qtcoinviewer.cpp:3401 _GetCameraImage] cannot render offscreen
[qtcoinviewer.cpp:3401 _GetCameraImage] cannot render offscreen
```

PeterBowman, 2017-12-07:
http://openrave-users-list.185357.n3.nabble.com/qtcoinviewer-failure-when-activating-camera-sensor-tp4027743p4027756.html:

> After a little of research and several tests, it looks like there is some problems when using the coin three party library on Ubuntu 14.04 64bits.
>
> I have set up everything on a new partition running the same Ubuntu 14.04 but this time the 32 bits version and everything is running ok.
>
> [...]
>
> I wonder if the problem is because of the three party libraries or due to the nvidia driver when using a 64 OS.

jgvictores, 2018-01-20:
This is replicable via [openraveYarpPluginLoader-grabber.py](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/b3fdf865d349f108d7fd00bf123a76d5cdcb9c86/examples/openraveYarpPluginLoader-grabber.py) which invokes the modern [YarpOpenraveGrabber](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/tree/b3fdf865d349f108d7fd00bf123a76d5cdcb9c86/libraries/YarpPlugins/YarpOpenraveGrabber),
 even with a working NVIDIA 304.135 linux driver.

jgvictores, 2018-01-20:
However, on same machine, via `yarpdev --device YarpOpenraveGrabber --env /usr/local/share/robotDevastation-openrave-models/contexts/openrave/ecro/mapping_room.env.xml --robotIndex 0 --sensorIndex 1 --view` now specified at [README.md](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/a5abdd8b75d05299c7d9572087ee22470c95bcee/libraries/YarpPlugins/YarpOpenraveGrabber/README.md), the old warning does not appear. The `yarpview` image is still black, but the only message is:
```
[WARNING]ServerFrameGrabber: error configuring interfaces for parsers 
```

jgvictores, 2018-01-20:
> However, on same machine, via:
> 
> yarpdev --device YarpOpenraveGrabber --env /usr/local/share/robotDevastation-openrave-models/contexts/openrave/ecro/mapping_room.env.xml --robotIndex 0 --sensorIndex 1 --view
> ...as now specified at README.md, the old cannot render offscreen warning does not appear. The yarpview image is still black, but the only message is:
> 
> [WARNING]ServerFrameGrabber: error configuring interfaces for parsers

Might actually be a race condition as described at #60: Via CLI, [OpenRAVE::SensorBase::CC_PowerOn](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/dd8c301b18d06074fc5cab36d132710f726f9553/libraries/YarpPlugins/YarpOpenraveRGBDSensor/DeviceDriverImpl.cpp#L70-L71) has no apparent effect. This is bad...

jgvictores, 2018-01-21:
`SoOffscreenRenderer`

jgvictores, 2018-01-21:
https://askubuntu.com/questions/745135/how-to-enable-indirect-glx-contexts-iglx-in-ubuntu-14-04-lts-with-nvidia-gfx/747088

jgvictores, 2018-01-21:
Not solved, but finally some progress here. Hers's what I've got so far (disclaimer: AFAIK):
1. OpenRAVE ultimately uses `SoOffscreenRenderer` for what it calls offscreen rendering, as can be seen in [plugins/qtcoinrave/qtcoinviewer.cpp](https://github.com/rdiankov/openrave/blob/master/plugins/qtcoinrave/qtcoinviewer.cpp#L3420).
1. I copied some code creating [exampleSoOffscreenRenderer](https://github.com/jgvictores/snippets/tree/315d2fe740447d21e586dc769efe394c37e45af7/coin3d) (origins stated at top of each file), which running on a Ubuntu Xenial returned what I put [above](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/48#issuecomment-359214421):
   ```
   Major opcode of failed request:  155 (GLX)
   Minor opcode of failed request:  3 (X_GLXCreateContext)
   ```
1. `SoOffscreenRenderer`, for offscreen rendering, uses something more generally called indirect GLX rendering.
1. Indirect GLX rendering has been disabled in modern distros due to security issues (sorry, I've lost the exact reference to this statement, but it was not official documentation at all, more like a forum archive).
1. From [this askubuntu question](https://askubuntu.com/questions/745135/how-to-enable-indirect-glx-contexts-iglx-in-ubuntu-14-04-lts-with-nvidia-gfx/747088) I stated [above](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/48#issuecomment-359216090) I saw you could enable it via a section in `/etc/X11/xorg.conf` (copying the one with two `option`s, seems more complete):
   ```
   Section "ServerFlags"  
      Option "AllowIndirectGLX" "on"  
      Option "IndirectGLX" "on"  
   EndSection
   ```
   This file no longer exists in Ubuntu Xenial, so I put that contents in a newly created `/usr/share/X11/xorg.conf.d/80-custom-glx.conf`.
1. Reboot, run [exampleSoOffscreenRenderer](https://github.com/jgvictores/snippets/tree/315d2fe740447d21e586dc769efe394c37e45af7/coin3d) (origins stated at top of each file) again, and error has changed to [ye oldie](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/48#issuecomment-349433838):
   ```
   Coin info in glglue_allow_newer_opengl(): 
   
   Features of OpenGL version > 1.0 has been
   disabled, due to the use of a remote display.
   
   This is so because many common OpenGL drivers
   have problems in this regard.
   
   To force full OpenGL use, set the environment
   variable COIN_FULL_INDIRECT_RENDERING=1 and re-run the application.
   
   If you don't want this message displayed again,
   set the environment variable COIN_DONT_INFORM_INDIRECT_RENDERING=1.
   ```
1. Run `COIN_FULL_INDIRECT_RENDERING=1 exampleSoOffscreenRenderer`. No more progress (same message saying to set `COIN_FULL_INDIRECT_RENDERING=1`) on a machine that only has an integrated Intel and no NVIDIA.
1. Replicate all the above (including the `80-custom-glx.conf` file) on the machine that has a correctly installed `NVIDIA 304.135` driver. All the same, except final step: running `COIN_FULL_INDIRECT_RENDERING=1 exampleSoOffscreenRenderer` leads to success! (well... there is a `Coin error in SbBool SoOffscreenRenderer::writeToFile(const SbString&, const SbName&) const(): simage library not available.`, but seems unrelated).
1. On same `NVIDIA 304.135` machine, ran `python openraveYarpPluginLoader-grabber.py`, getting the same message saying to set `COIN_FULL_INDIRECT_RENDERING=1`).
1. On same `NVIDIA 304.135` machine, ran `COIN_FULL_INDIRECT_RENDERING=1 python openraveYarpPluginLoader-grabber.py`, get no warning, but YARP sending black image.
1. On same `NVIDIA 304.135` machine, ran `COIN_FULL_INDIRECT_RENDERING=1 python openraveYarpPluginLoader-rgbdsensor.py`, get no warning, YARP still sending black image, but... and this is due to the fact that `YarpOpenraveRGBDSensor` [currently sends CC_RenderDataOn to the camera](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/dd8c301b18d06074fc5cab36d132710f726f9553/libraries/YarpPlugins/YarpOpenraveRGBDSensor/DeviceDriverImpl.cpp#L75)... got tiny window on screen with the image that is received from the camera!! :tada::tada::tada:

Again, a black image is still published via YARP, but the tiny window on screen with the image that is received from the camera was the behavior back when everything worked (this is, the desired behavior of setting `CC_RenderDataOn`). It looks like it means offscreen rendering is working, and now just have to check our own implementation of getting the offscreen rendered image through to YARP.

jgvictores, 2018-01-21:
> Coin error in SbBool SoOffscreenRenderer::writeToFile(const SbString&, const SbName&) const(): simage library not available

For the record, this was solved via `sudo apt install libsimage-dev`, example now finally generating the pronised 5 `.jpg` files.

jgvictores, 2018-01-21:
> `sudo apt install libsimage-dev`

Did not help with this issue. :-)

jgvictores, 2018-01-21:
Guys, we are back on track! :tada::tada::tada::tada::tada:

Decided to go back in time to `testRaveBot`:
```bash
COIN_DONT_INFORM_INDIRECT_RENDERING=1 testRaveBot --env /usr/local/share/asibot/contexts/openrave/models/asibot_kitchen_cameras.env.xml
```
[image]

jgvictores, 2018-01-21:
Fixed command above (update to `COIN_FULL_INDIRECT_RENDERING=1`), removed `upstream` and added `epic` label. :-)

jgvictores, 2018-01-21:
`YarpOpenraveGrabber` fixed at a7d73f6e4dc74144a1b601fda3b93f709a584abc, mostly thanks to d04eb26a9e4f7777f2afa7fb77286b0d9bb4c64b. Remaining work on `YarpOpenraveRGBDSensor` will be tracked on #58 as its title indicates.

- Bad news on CLI side: it does seem we are affected by #60, specifically see [this comment](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/60#issuecomment-359206462).
- Good news on the Python side (launching with `COIN_FULL_INDIRECT_RENDERING=1 python` on the left):
   - [examples/openraveYarpPluginLoader-grabber.py](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/a7d73f6e4dc74144a1b601fda3b93f709a584abc/examples/openraveYarpPluginLoader-grabber.py) has been adapted to load `asibot` or `ecro` as streaming simulated camera examples.
   - [examples/openraveYarpPluginLoader-robotmanager.py](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/a7d73f6e4dc74144a1b601fda3b93f709a584abc/examples/openraveYarpPluginLoader-robotmanager.py) is now fully compliant and has been tested with Robot Devastation (bump: https://github.com/asrob-uc3m/robotDevastation-robots/issues/36)! :tada:

Closing issue, and will begin documenting where required.

closed 2018-01-21 by jgvictores

jgvictores, 2018-01-21:
More literature on this topic:
- https://www.phoronix.com/scan.php?page=news_item&px=Xorg-IGLX-Potential-Bye-Bye
- https://cgit.freedesktop.org/xorg/xserver/commit/?id=4fca18dc03669bada75175df289fbee41275208e&utm_source=anzwix

jgvictores, 2018-01-23:
Exact process replicated on machine with `GeForce GTX 675M` with driver `NVIDIA 384.111`. :tada::tada::tada:

Only new discovery, which actually matches behavior observed on the other machine: `COIN_FULL_INDIRECT_RENDERING=1` is not required at all for this, has no real or visible effect. In practice, has the same effect as `COIN_DONT_INFORM_INDIRECT_RENDERING=1`: omitting the warning. :-)

jgvictores, 2018-01-24:
Other machine was `GeForce GTX 260` with `NVIDIA 304.135`.

Same current uselessness of `COIN_FULL_INDIRECT_RENDERING=1` observed.

jgvictores, 2018-01-24:
BTW, by `(proprietary) drivers correctly installed` I mean GPU shows on `nvidia-settings`, with complete information.

Other notes:
- Vanilla open-source `nouveau` does not seem to be enough, at least for NVIDIA.
- Old records indicate that open-source `bumblebee/optirun` were good enough for offscreen rendering on the `GeForce GTX 675M` machine.

PeterBowman, 2018-09-27:
Under the assumption that a PC has updated NVIDIA drivers and OR offscreen rendering works, i.e. it can display grabbed frames when using a desktop, the corresponding YARP camera port sends empty bottles when the simulation was run on said PC over SSH/X11 (`ssh -X remote`) without a viewer. With `env.SetViewer('qtcoin')`:

```
$ python openraveYarpPluginLoader-grabber.py 
[plugindatabase.h:645 Create] Failed to create name fcl_, interface collisionchecker
X Error: BadValue (integer parameter out of range for operation) 2
  Extension:    155 (Uknown extension)
  Minor opcode: 3 (Unknown request)
  Resource id:  0x0
terminate called after throwing an instance of 'OpenRAVE::openrave_exception'
  what():  openrave (Assert): [/usr/include/boost/thread/pthread/mutex.hpp:111] -> boost::mutex::~mutex(), expr: !res
```

And `yarpview` crashes with a similar error message:

```
$ yarpview
yarp: Port /yarpview/img:i active at tcp://163.117.150.242:10004/
Failed to create OpenGL context for format QSurfaceFormat(version 2.0, options QFlags(), depthBufferSize 24, redBufferSize -1, greenBufferSize -1, blueBufferSize -1, alphaBufferSize -1, stencilBufferSize 8, samples -1, swapBehavior 2, swapInterval 1, profile  0) 
Aborted (core dumped)
```

Same results over regular SSH (no `-X` option), with and without a viewer.

PeterBowman, 2018-09-28:
> For the record, this was solved via `sudo apt install libsimage-dev`, example now finally generating the promised 5 `.jpg` files.

I confirm this is not necessary in order to make offscreen rendering work, it was just needed for the sample program to save rendered frames to disk.

Hopefully, all you need is a (modern?) NVIDIA GPU, its proprietary drivers, `/usr/share/X11/xorg.conf.d/80-custom-glx.conf` as detailed in https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/48#issuecomment-359241486, and a reboot.

PeterBowman, 2018-09-28:
https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/48#issuecomment-425073153

That was the output I got while trying to run the simulation over SSH **from a PC with AMD graphics**. The other way around (starting it on an AMD machine over SSH while connecting from a PC with fully configured NVIDIA graphics) I get this:

```
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
terminate called after throwing an instance of 'OpenRAVE::openrave_exception'
  what():  openrave (Assert): [/usr/include/boost/thread/pthread/mutex.hpp:111] -> boost::mutex::~mutex(), expr: !res
```

Also, disclaimer: I can't render offscreen if OpenRAVE's viewer has not been started (locally).

jgvictores, 2018-11-09:
I just ran into [this interesting guide](https://www.scm.com/doc/Installation/Remote_GUI.html) (because I was looking for info on indirect rendering on AMD), and it touches some topics that may be useful.
For instance, our `/usr/share/lightdm/lightdm.conf.d/50-xserver-command.conf` reads:
```
[Seat:*]
# Dump core
xserver-command=X -core
```
This is, no `+iglx` for now (or anywhere in titanx's `/usr/share/lightdm/` from what I've seen).

PeterBowman, 2018-11-16:
I could not run `glxgears` until I did [this trick](https://www.scm.com/doc/Installation/Remote_GUI.html#intel-graphics-mesa). Prior to that, `glxinfo |  grep -E " version| string| rendering|display` was showing:

`direct rendering: No (If you want to find out why, try setting LIBGL_DEBUG=verbose)`

Now, `python openraveYarpPluginLoader-grabber.py` is successful (it starts an OR instance with its viewer), but `[qtcoinviewer.cpp:3405 _GetCameraImage] cannot render offscreen` warnings are constantly popping up.

You may noticed I linked to the Intel graphics section. My laptop has actually two GPUs: integrated Intel and dedicated AMD.

jgvictores, 2018-11-19:
Just to link with an OR issue that blames `SoOffscreenRenderer` as we do above: https://github.com/rdiankov/openrave/issues/238#issuecomment-10777959

jgvictores, 2018-11-19:
Just some more links (BTW, this is mostly because I currently have no local machine with NVIDIA :joy:):
- http://wiki.compiz.org/Intel%20with%20AiGLX
- https://wiki.gentoo.org/wiki/Xorg/Hardware_3D_acceleration_guide#What_is_DRI.3F

jgvictores, 2019-12-12:
BTW, been trying with OpenSceneGraph (OSG deserves a full issue; you need https://github.com/openscenegraph/OpenSceneGraph/issues/212#issuecomment-432872511 and set `-DDESIRED_QT_VERSION=4` as mentioned at https://scaron.info/teaching/installing-openrave-on-ubuntu-16.04.html)... seems like no camera at all (even with NVIDIA). Notory line: `2019-12-12 11:06:32,165 openrave [WARN] [qtosgviewer.cpp:279 QtOSGViewer::customEvent] got a qt message that isn't of MyCallbackEvent, converting statically (dangerous)
`

jgvictores, 2019-12-15:
Info added at https://github.com/roboticslab-uc3m/installation-guides/commit/bf7a36b513bc11019d8535105c9d844f1c1ee7b4

jgvictores, 2019-12-16:
> 2019-12-12 11:06:32,165 openrave [WARN] [qtosgviewer.cpp:279 QtOSGViewer::customEvent] got a qt message that isn't of MyCallbackEvent, converting statically (dangerous

https://github.com/rdiankov/openrave/blob/v0.9.0/plugins/qtosgrave/qtosgviewer.cpp#L279

jgvictores, 2019-12-16:
> https://github.com/rdiankov/openrave/blob/v0.9.0/plugins/qtosgrave/qtosgviewer.cpp#L279

Continuing with this slightly offtopic, just to confirm OSG is not the way to get offscreen render without NVIDIA:
- Commands registered by qtcoin as of `v0.9.0`: https://github.com/rdiankov/openrave/blob/v0.9.0/plugins/qtcoinrave/qtcoinviewer.cpp#L180-L201
- Commands registered by qtosg as of `v0.9.0`: https://github.com/rdiankov/openrave/blob/v0.9.0/plugins/qtosgrave/qtosgviewer.cpp#L143-L160
- Commands registered by qtosg as of current `master`: https://github.com/rdiankov/openrave/blob/abe85928e74381d3d3d33db73693d92c9fbd7a1c/plugins/qtosgrave/qtosgviewer.cpp#L144-L165

jgvictores, 2019-12-23:
> Continuing with this slightly offtopic, just to confirm OSG is not the way to get offscreen render without NVIDIA:

Furthermore, I'm getting a segfault with a simple `openrave data/testwamcamera.env.xml --viewer qtosg`.

jgvictores, 2019-12-23:
Even more reads:
- https://pyrender.readthedocs.io/en/latest/examples/offscreen.html
- https://github.com/personalrobotics/offscreen_render (thanks https://github.com/crigroup/robotiq/issues/2 for `git clone https://github.com/crigroup/openrave_catkin`)
    ```bash
    [rospack] Error: package 'offscreen_render' not found
    [librospack]: error while executing command
    Failed to open /shaders/VertexShader.glsl
    ```
- https://github.com/openai/mujoco-py/issues/408

jgvictores, 2019-12-23:
> ```shell
> [rospack] Error: package 'offscreen_render' not found
> ```

Fixex via `source devel/setup.bash`. Still, no progress (blank image, tested on machine without NVIDIA).

jgvictores, 2020-01-10:
- `export LIBGL_ALWAYS_SOFTWARE=1` does not do much.
- `export LIBGL_DEBUG=verbose`. We discover `swrast_dri.so` (good). Also `/home/yo/.drirc` not present, but Xenial seems to use `/etc/drirc`. Also interesting `xdriinfo options 0` and https://dri.freedesktop.org/wiki/ConfigurationInfrastructure/