roboticslab-uc3m/openrave-yarp-plugins #54: Move FakeControlBoard to yarp-devices

PeterBowman, 2017-12-17:
The [FakeControlBoard device](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/tree/44296ec7cfadf7f6cbe4a316ce0cbd224b0056a5/libraries/YarpPlugins/FakeControlboard) actually belongs to [roboticslab-uc3m/yarp-devices](https://github.com/roboticslab-uc3m/yarp-devices). We only needed this here because of teoSim:

https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/44296ec7cfadf7f6cbe4a316ce0cbd224b0056a5/programs/teoSim/ControlboardContainer.cpp#L66

...which is about to die as a result of #27.

When completed, remember to update unit tests at kinematics-dynamics ([.travis.yaml](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/5f0383af9e5c13fecac8e45e4e9b5bc30e4ac288/.travis.yml#L34) + [doc/kinematics-dynamics-install.md](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/5f0383af9e5c13fecac8e45e4e9b5bc30e4ac288/doc/kinematics-dynamics-install.md#kinematics-dynamics-installation-from-source-code)).

This issue is preceded by 8b4a803 (code & style review, minor refactorization of FCB).

jgvictores, 2017-12-20:
Given that it doesn't even depend on ColorDebug, maybe we could clean it up a bit and try a PR at YARP?

jgvictores, 2017-12-20:
PS: Oh, wait! It does depend on ColorDebug, see [here](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/44296ec7cfadf7f6cbe4a316ce0cbd224b0056a5/libraries/YarpPlugins/FakeControlboard/DeviceDriverImpl.cpp#L8).

PeterBowman, 2017-12-20:
> PS: Oh, wait! It does depend on ColorDebug, see here.

My bad: 4dfa391 :(. We can revert it, though.

> maybe we could clean it up a bit and try a PR at YARP?

For reference: http://yarp.it/classyarp_1_1dev_1_1TestMotor.html.

PeterBowman, 2017-12-21:
Thanks! I tweaked our unit test at kinematics-dynamics to use *fakeMotionControl* (in [here](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/5f0383af9e5c13fecac8e45e4e9b5bc30e4ac288/test/testBasicCartesianControl.cpp#L26), passing a compulsory `name` option as well), but kept being unable to acquire the `yarp::dev::IVelocityControl` interface handle (crashes [here](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/5f0383af9e5c13fecac8e45e4e9b5bc30e4ac288/libraries/YarpPlugins/BasicCartesianControl/DeviceDriverImpl.cpp#L46)). It does not happen when I upgrade the velocity interface to `yarp::dev::IVelocityControl2` ([ref](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/5f0383af9e5c13fecac8e45e4e9b5bc30e4ac288/libraries/YarpPlugins/BasicCartesianControl/BasicCartesianControl.hpp#L189)).

jgvictores, 2017-12-21:
@traversaro Thanks for the reference!
@PeterBowman Looks good! I put https://github.com/robotology/yarp/issues/1502 just to keep thing in place. :-)

PeterBowman, 2018-01-05:
We observed that `fakeMotionControl` does not emulate velocities in position or velocity control mode.
For now, waiting for #27, and then will move `FakeControlboard` to yarp-devices.

jgvictores, 2018-01-12:
No longer blocked!

PeterBowman, 2018-01-13:
Done at:

* 9289d7e
* [roboticslab-uc3m/yarp-devices@fafe871...18aa6ab](https://github.com/roboticslab-uc3m/yarp-devices/compare/fafe871...18aa6ab)
* roboticslab-uc3m/kinematics-dynamics@aa9a20c

closed 2018-01-13 by PeterBowman

jgvictores, 2018-01-13:
Thank you so much!!