roboticslab-uc3m/openrave-yarp-plugins #55: Think about implementing additional motor/axis yarp::dev interfaces

jgvictores, 2017-12-22:
Think about implementing additional motor/axis `yarp::dev` interfaces:
- http://www.yarp.it/classyarp_1_1dev_1_1IMotor.html
- http://www.yarp.it/classyarp_1_1dev_1_1IMotorEncoders.html
- http://www.yarp.it/classyarp_1_1dev_1_1IAxisInfo.html

PeterBowman, 2019-07-29:
> http://www.yarp.it/classyarp_1_1dev_1_1IAxisInfo.html

Might be blocked by https://github.com/roboticslab-uc3m/teo-main/issues/44 (renaming joints).

jgvictores, 2019-11-19:
> http://www.yarp.it/classyarp_1_1dev_1_1IAxisInfo.html

Done at https://github.com/roboticslab-uc3m/openrave-yarp-plugins/commit/571abb9d8c6b987c9d25fcca01f82113c4cde425

> Might be blocked by roboticslab-uc3m/teo-main#44 (renaming joints).

Thanks to https://github.com/roboticslab-uc3m/teo-openrave-models/pull/35 for https://github.com/roboticslab-uc3m/teo-main/issues/44, we actually have the pretty names. ^^

[image]

PeterBowman, 2023-08-28:
Besides, we are missing an implementation for `IPositionControl::getRefAcceleration` and `IEncoders::getEncoderAcceleration` (at least a dummy one which simply returns zero and true). It is causing trouble in:
* https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/f4bc01f1325efc64317be58e4eeb3fc44f27f496/libraries/YarpPlugins/BasicCartesianControl/PeriodicThreadImpl.cpp#L36
* https://github.com/roboticslab-uc3m/teo-self-presentation/blob/74a6ff90b28bd12637ec2c3828d1277b659fac57/programs/BodyExecution/BodyExecution.cpp#L79-L83