roboticslab-uc3m/openrave-yarp-plugins #56: Refactor CMake files

PeterBowman, 2017-12-24:
Much like roboticslab-uc3m/kinematics-dynamics#14 and roboticslab-uc3m/yarp-devices#26.

PeterBowman, 2017-12-24:
Marking as blocked for the time being to not interfere with ongoing work.

PeterBowman, 2018-01-17:
Almost done at the `fix-56-cmake` branch, I'm still waiting for Travis to do its job.

PeterBowman, 2018-01-17:
Merged in 114daaf. Now, CMake 2.8.11 or later is required. Apart from the usual cleanup, the [`InstallOpenravePlugin`](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/114daaf2e852285caff5f0eabc92a3e523ca6718/cmake/modules/InstallOpenravePlugin.cmake) CMake module has been revamped and is now available for use by clients via `find_package` (see [`ROBOTICSLAB_OPENRAVE_YARP_PLUGINSConfig.cmake.in`](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/114daaf2e852285caff5f0eabc92a3e523ca6718/cmake/template/ROBOTICSLAB_OPENRAVE_YARP_PLUGINSConfig.cmake.in))

closed 2018-01-17 by PeterBowman

jgvictores, 2018-01-17:
Looks really good! @PeterBowman Thanks a lot for these improvements!!