roboticslab-uc3m/openrave-yarp-plugins #59: OpenraveYarpPluginLoader via `main()`: plugin destructor not called

jgvictores, 2017-12-26:
This issue is a follow-up on https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/47#issuecomment-353725800.

OpenraveYarpPluginLoader via `main()`: plugin destructor not called, for example when exiting via `^C` after calling:
```bash
openrave /usr/local/share/teo-openrave-models/contexts/openrave/teo/teo.robot.xml --module OpenraveYarpPluginLoader "open --device controlboardwrapper2 --subdevice YarpOpenraveControlboard --robotIndex 0 --allManipulators"
```

Destructors are called in Python scripts, but note that there [`RaveDestroy()` is called explicity](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/3e21fa527ea328d08375b0ebcbd62a43c2b31b34/examples/openraveYarpPluginLoader-controlboard.py#L19-L23).

PeterBowman, 2017-12-26:
There is a signal handler in the `openrave` CLI utility and it *did* call `RaveDestroy()` ([ref](https://github.com/rdiankov/openrave/blob/b2818fb534cd8c4e64ea1dddea0f7c360ce5056d/src/openrave.cpp#L386-L404)). However, it was removed in rdiankov/openrave@cbfd27a for the sake of *stability to osg viewer*.

PeterBowman, 2017-12-26:
The destructor does get called when I restore `RaveDestroy();` and `s_penv.reset();` in the signal handler. Perhaps we should file an issue upstream. Meanwhile, I'd suggest closing the viewer window by clicking on the corresponding button, which will indeed call the destructor regardless of those two commented lines.

---

Stack trace of `OpenraveYarpPluginLoader`'s constructor for debugging purposes (upstream at rdiankov/openrave@b2818fb):

```
Thread #1 [openrave0.9] 13837 [core: 2] (Suspended : Breakpoint)	
	OpenraveYarpPluginLoader::OpenraveYarpPluginLoader() at OpenraveYarpPluginLoader.cpp:42 0x7fffce4437ca	
	CreateInterfaceValidated() at OpenraveYarpPluginLoader.cpp:351 0x7fffce440d64	
	OpenRAVECreateInterface() at plugin.h:75 0x7fffce440634	
	OpenRAVE::RaveDatabase::Plugin::CreateInterface() at plugindatabase.h:292 0x7fffeef882ec	
	OpenRAVE::RaveDatabase::Create() at plugindatabase.h:602 0x7fffeef8b715	
	OpenRAVE::RaveDatabase::CreateModule() at plugindatabase.h:407 0x7fffeef897b5	
	OpenRAVE::RaveCreateModule() at libopenrave.cpp:1,249 0x7fffeef72f08	
	MainOpenRAVEThread() at openrave.cpp:350 0x42d6e8	
	main() at openrave.cpp:280 0x42c9de	
```

jgvictores, 2018-01-08:
Yes, closing the viewer window by clicking on the corresponding button works for me too.

BTW, a hack `yarpmanager` used to have was to catch the CTRL-C signal and simply display a message suggest closing the viewer window by clicking on the corresponding button.