roboticslab-uc3m/openrave-yarp-plugins #61: openrave-yarp-plugins examples (iron and paint) failing to load controlBoardWrapper2

RaulFdzbis, 2017-12-28:
So when i execute both examples i get the following errors from console. 

## Paint

```bash
2017-12-28 18:36:12,385 openrave [WARN] [plugindatabase.h:577 InterfaceBasePtr OpenRAVE::RaveDatabase::Create] Failed to create name fcl_, interface collisionchecker
[info] OpenraveYarpPluginLoader.cpp:112 Open(): Checking for yarp network...
[success] OpenraveYarpPluginLoader.cpp:118 Open(): Found yarp network.
[debug] OpenraveYarpPluginLoader.cpp:142 Open(): config: (robotIndex 0) (manipulatorIndex 2) (genRefSpeed 999999) (device controlboardwrapper2) (subdevice YarpOpenraveControlboard)
[info] OpenraveYarpPluginLoader.cpp:145 Open(): penv: 0x1cfd730
[DEBUG]ControlBoardWrapper2: 'period' parameter missing, using default thread period = 20ms 
[DEBUG]opening controlBoardWrapper2 subdevice
error while opening /usr/local/lib/yarp/YarpOpenraveControlboard.so:
  libYarpOpenraveBase.so: cannot open shared object file: No such file or directory
Cannot load plugin from shared library (YarpOpenraveControlboard)
(YarpOpenraveControlboard: cannot open shared object file: No such file or directory)
yarpdev: ***ERROR*** could not find device <YarpOpenraveControlboard>
[ERROR]opening controlBoardWrapper2 subdevice... FAILED
[ERROR]ControlBoardWrapper: error while opening subdevice
yarpdev: ***ERROR*** driver <controlboardwrapper2> was found but could not open
[error] OpenraveYarpPluginLoader.cpp:294 Open(): yarp plugin not valid.
None
[info] OpenraveYarpPaintSquares.cpp:137 Open(): Checking for yarp network...
[success] OpenraveYarpPaintSquares.cpp:143 Open(): Found yarp network.
[debug] OpenraveYarpPaintSquares.cpp:167 Open(): config: (squares 4)
[info] OpenraveYarpPaintSquares.cpp:170 Open(): port name: /openraveYarpPaintSquares/rpc:s
[info] OpenraveYarpPaintSquares.cpp:173 Open(): squares: 4
2017-12-28 18:36:13,783 openrave [INFO] [OpenraveYarpPaintSquares.cpp:175 Open] penv: 0x1cfd730
sucess: object "object" exists.
sucess: object "wall" exists.
Robot 0: teoSim
yarp: Port /openraveYarpPaintSquares/rpc:s active at tcp://192.168.0.160:10003/
```

## Iron

```bash
2017-12-28 18:37:16,358 openrave [WARN] [plugindatabase.h:577 InterfaceBasePtr OpenRAVE::RaveDatabase::Create] Failed to create name fcl_, interface collisionchecker
2017-12-28 18:37:17,150 openrave [WARN] [xmlreaders-core.cpp:2591 ManipulatorXMLReader::endElement] failed to create iksolver ik/ikfast0x10000048.Transform6D.x86_64.16_17_18_19_20_21.so
2017-12-28 18:37:17,150 openrave [WARN] [xmlreaders-core.cpp:2591 ManipulatorXMLReader::endElement] failed to create iksolver ik/ikfast0x10000048.Transform6D.x86_64.22_23_24_25_26_27.so
2017-12-28 18:37:17,150 openrave [WARN] [plugindatabase.h:577 InterfaceBasePtr OpenRAVE::RaveDatabase::Create] Failed to create name forcesensor, interface sensor
2017-12-28 18:37:17,150 openrave [ERROR] [xmlreaders-core.cpp:1797 InterfaceXMLReader::InterfaceXMLReader] xml readers failed to create instance of type sensor:forcesensor
2017-12-28 18:37:17,150 openrave [WARN] [plugindatabase.h:492 InterfaceBasePtr OpenRAVE::RaveDatabase::Create] interface sensor name "" needs to start with a valid character
2017-12-28 18:37:17,151 openrave [ERROR] [xmlreaders-core.cpp:1835 InterfaceXMLReader::_CheckInterface] xml readers failed to create instance of type sensors
2017-12-28 18:37:17,151 openrave [WARN] [plugindatabase.h:492 InterfaceBasePtr OpenRAVE::RaveDatabase::Create] interface sensor name "" needs to start with a valid character
2017-12-28 18:37:17,151 openrave [ERROR] [xmlreaders-core.cpp:1835 InterfaceXMLReader::_CheckInterface] xml readers failed to create instance of type sensors
2017-12-28 18:37:17,654 openrave [WARN] [environment-core.h:487 Load] load failed on file /home/raul/repos/textiles/textiles/ironing/manipulation/ironingSim/ironingSim.env.xml
[info] OpenraveYarpPluginLoader.cpp:112 Open(): Checking for yarp network...
[success] OpenraveYarpPluginLoader.cpp:118 Open(): Found yarp network.
[debug] OpenraveYarpPluginLoader.cpp:142 Open(): config: (robotIndex 0) (manipulatorIndex 2) (genRefSpeed 999999) (device controlboardwrapper2) (subdevice YarpOpenraveControlboard)
[info] OpenraveYarpPluginLoader.cpp:145 Open(): penv: 0x1a7e4b0
[DEBUG]ControlBoardWrapper2: 'period' parameter missing, using default thread period = 20ms 
[DEBUG]opening controlBoardWrapper2 subdevice
error while opening /usr/local/lib/yarp/YarpOpenraveControlboard.so:
  libYarpOpenraveBase.so: cannot open shared object file: No such file or directory
Cannot load plugin from shared library (YarpOpenraveControlboard)
(YarpOpenraveControlboard: cannot open shared object file: No such file or directory)
yarpdev: ***ERROR*** could not find device <YarpOpenraveControlboard>
[ERROR]opening controlBoardWrapper2 subdevice... FAILED
[ERROR]ControlBoardWrapper: error while opening subdevice
yarpdev: ***ERROR*** driver <controlboardwrapper2> was found but could not open
[error] OpenraveYarpPluginLoader.cpp:294 Open(): yarp plugin not valid.
None
[info] OpenraveYarpPluginLoader.cpp:112 Open(): Checking for yarp network...
[success] OpenraveYarpPluginLoader.cpp:118 Open(): Found yarp network.
[debug] OpenraveYarpPluginLoader.cpp:142 Open(): config: (robotIndex 0) (manipulatorIndex 0) (genRefSpeed 999999) (device controlboardwrapper2) (subdevice YarpOpenraveControlboard)
[info] OpenraveYarpPluginLoader.cpp:145 Open(): penv: 0x1a7e4b0
[DEBUG]ControlBoardWrapper2: 'period' parameter missing, using default thread period = 20ms 
[DEBUG]opening controlBoardWrapper2 subdevice
error while opening /usr/local/lib/yarp/YarpOpenraveControlboard.so:
  libYarpOpenraveBase.so: cannot open shared object file: No such file or directory
Cannot load plugin from shared library (YarpOpenraveControlboard)
(YarpOpenraveControlboard: cannot open shared object file: No such file or directory)
yarpdev: ***ERROR*** could not find device <YarpOpenraveControlboard>
[ERROR]opening controlBoardWrapper2 subdevice... FAILED
[ERROR]ControlBoardWrapper: error while opening subdevice
yarpdev: ***ERROR*** driver <controlboardwrapper2> was found but could not open
[error] OpenraveYarpPluginLoader.cpp:294 Open(): yarp plugin not valid.
None
2017-12-28 18:37:17,705 openrave [INFO] [OpenraveYarpForceEstimator.cpp:138 Open] portName: /openraveYarpForceEstimator/rpc:s
yarp: Port /openraveYarpForceEstimator/rpc:s active at tcp://192.168.0.160:10002/
2017-12-28 18:37:17,737 openrave [INFO] [OpenraveYarpForceEstimator.cpp:147 Open] penv: 0x1a7e4b0
Robot 0: teoSim
2017-12-28 18:37:17,778 openrave [WARN] [qtcoinrave.cpp:80 DestroyPlugin] SoQt releasing all memory
Coin debug: socontexthandler_cleanup(): 4 context-bound resources not free'd before exit.
Traceback (most recent call last):
  File "openraveYarpForceEstimator.py", line 27, in <module>
    print OpenraveYarpForceEstimator.SendCommand('open')
openravepy._openravepy_.openravepy_ext.openrave_exception: openrave (InvalidArguments): [virtual bool OpenRAVE::RobotBase::Grab(OpenRAVE::KinBodyPtr, OpenRAVE::KinBody::LinkPtr):1480] (!!pbody) failed robot teoSim invalid bod to grab
destroying plugin
Segmentation fault (core dumped)
```

## Important Notes:

- I am using a different computer than the one i usually use, so having a problem with yarp/openrave installation is not discarded.
- In the paint action the visualization is correctly loaded, in IRON the visualization is **NOT** loaded and crush before initialization. This may be a **different** problem.

- If I remember correctly xmlreader errors are not something new with iron, they have appeared before, but didnt make the system to fail.

jgvictores, 2017-12-28:
Could you update like other examples? 
Basically it's the transition from YarpOpenraveControlboard to YarpOpenravePluginLoader. If you link to the code I could probably give you the exact format.

PeterBowman, 2017-12-28:
Should [YarpOpenraveBase](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/tree/9e29a31d79a808f78a966bdd27cc7a8b326b3bb8/libraries/YarpPlugins/YarpOpenraveBase) live in the `YarpPlugins/` directory in spite of not being a YARP plugin?

RaulFdzbis, 2017-12-29:
>Could you update like other examples?
Basically it's the transition from YarpOpenraveControlboard to YarpOpenravePluginLoader. If you link to the code I could probably give you the exact format.

I think that this transition is already done. Check [code]( https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/9e29a31d79a808f78a966bdd27cc7a8b326b3bb8/examples/openraveYarpForceEstimator.py#L19).

jgvictores, 2017-12-29:
Could you try a `sudo ldconfig`?

RaulFdzbis, 2017-12-29:
>Could you try a sudo ldconfig?

Yes that was it. Probably i forgot to do that one of all the times i had to install because of conda. Thank you @jgvictores !!!

The iron example still failing though:

```bash
Traceback (most recent call last):
  File "openraveYarpForceEstimator.py", line 27, in <module>
    print OpenraveYarpForceEstimator.SendCommand('open')
openravepy._openravepy_.openravepy_ext.openrave_exception: openrave (InvalidArguments): [virtual bool OpenRAVE::RobotBase::Grab(OpenRAVE::KinBodyPtr, OpenRAVE::KinBody::LinkPtr):1480] (!!pbody) failed robot teoSim invalid bod to grab
destroying plugin
Segmentation fault (core dumped)
```

I will try to do some changes, and if it still no working, i will open a proper issue.

jgvictores, 2017-12-29:
Iron does a Grab that fails if the object to grab is not in the environment. Please extra-check the name of the grabbed object and that it exists in the environment.

RaulFdzbis, 2017-12-29:
Solved. The problem was that i was using the wrong textiles branch, I had the [develop](https://github.com/roboticslab-uc3m/textiles/tree/develop) branch, while for these examples the [textiles-xgnitive](https://github.com/roboticslab-uc3m/textiles/tree/textiles-xgnitive) is the needed one.

The reason of this is because some renaming in the model between branches, giving the error you already mention. Thanks for the help!. I think this can be closed.

closed 2017-12-29 by RaulFdzbis