roboticslab-uc3m/openrave-yarp-plugins #62: Should YarpOpenraveBase live in YarpPlugins directory?

jgvictores, 2018-01-02:
From https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/61#issuecomment-354334328
> Should [YarpOpenraveBase](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/tree/9e29a31d79a808f78a966bdd27cc7a8b326b3bb8/libraries/YarpPlugins/YarpOpenraveBase) live in the `YarpPlugins/` directory in spite of not being a YARP plugin (i.e. inheritance from `DeviceDriver` and such)?

PeterBowman, 2018-01-05:
Idea: use object libraries ([CMake docs](https://cmake.org/cmake/help/v3.5/command/add_library.html#object-libraries)) and pull up `YarpOpenraveBase`'s directory (place it as a sibling in `libraries/`).

jgvictores, 2018-01-05:
Blocked by #56

PeterBowman, 2018-01-17:
Done at c3bc1fd and merged into `develop`.

closed 2018-01-17 by PeterBowman

jgvictores, 2018-01-17:
Will try out very soon! Thanks a lot for this!!

PeterBowman, 2018-01-27:
Follow-up: object targets (at least in earlier CMake) cannot be imported, exported, installed nor linked. We noticed that Travis builds pulling YARP devel branch were failing due to the lack of support for C++11 features, that is, YARP 2.3.72 require that downstreams are compiled with C++11 options switched on. On usual static/shared libraries, usage requirements encapsulated within YARP targets propagate with each `target_link_libraries()` call. Since we are not able to use this due to limitations of object targets, the YarpOpenraveBase library failed to build due to missing mandatory compiler flags. I sorted this out at [428cbdf...d5a7944](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/compare/428cbdf...d5a7944), but better/cleaner ways might be found. Especially, I recall seeing that newer CMake releases changed the way object libraries are treated (see docs: [v3.0](https://cmake.org/cmake/help/v3.0/manual/cmake-buildsystem.7.html#id9) vs [latest](https://cmake.org/cmake/help/latest/manual/cmake-buildsystem.7.html#object-libraries)).

PeterBowman, 2018-01-27:
Interesting:

> Although object libraries may not be named directly in calls to the target_link_libraries() command, they can be “linked” indirectly by using an Interface Library whose INTERFACE_SOURCES target property is set to name $<TARGET_OBJECTS:objlib>.

[INTERFACE_SOURCES](https://cmake.org/cmake/help/v3.1/prop_tgt/INTERFACE_SOURCES.html) is available since CMake 3.1. Related (interface targets): roboticslab-uc3m/questions-and-answers#44.

PeterBowman, 2018-01-27:
I don't think the current approach with object targets can be improved beyond 1ecc801. In order to build YarpOpenraveBase with YARP's usage requirements (C++ features), the following lines seem mandatory:

https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/1ecc8018fb4a836300e896536398624c40888ae6/libraries/YarpOpenraveBase/CMakeLists.txt#L19-L27

What *could* be revamped a bit is how YarpOpenraveBase's own usage requirements are propagated, which looks like this right now:

```cmake
include_directories($<TARGET_PROPERTY:YarpOpenraveBase,INTERFACE_INCLUDE_DIRECTORIES>)

add_library(myLib myLib.hpp
                  myLib.cpp
                  $<TARGET_OBJECTS:YarpOpenraveBase>)
```

With interface libraries, in YarpOpenraveBase's `CMakeLists.txt`:

```cmake
add_library(YarpOpenraveBaseInterface INTERFACE)
target_include_directories(YarpOpenraveBaseInterface INTERFACE ${CMAKE_CURRENT_SOURCE_DIR})
set_target_properties(YarpOpenraveBaseInterface PROPERTIES INTERFACE_SOURCES $<TARGET_OBJECTS:YarpOpenraveBase>)
```

And in downstreams:

```cmake
add_library(myLib myLib.hpp
                  myLib.cpp)

target_link_libraries(myLib YarpOpenraveBaseInterface)
```

It looks like a matter of taste, though. Setting compiler options is more stringent and what `target_link_libraries()` can do for us is ATM handled manually. In fact, we are entirely ignoring potential requirements set by other YarpOpenraveBase's dependencies (YARP::YARP_dev, perhaps Boost...). The only robust solution I see here involves migrating back to (installable) shared libraries, or static ones (not tested).

PeterBowman, 2018-08-02:
See d0ad21f regarding the proposed `INTERFACE` target. We could go a bit further and refactor OpenRAVE's compiler and linker flags into another such target, which could also suit the `libraries/OpenravePlugins/` source tree. Anyway, CMake code is written using an explicit style in the current state of the repo, which is good. In the future, we could take advantage of CMake's improved support for `OBJECT` targets: configurable usage requirements via `target_link_libraries()`, installable targets, etc.

PeterBowman, 2019-12-15:
> In the future, we could take advantage of CMake's improved support for `OBJECT` targets: configurable usage requirements via `target_link_libraries()`, installable targets, etc.

This is happening now: https://github.com/roboticslab-uc3m/openrave-yarp-plugins/commit/07a15eca749513d9fc97bbb6026299f6191ac612. Further reading: [Linking Object Libraries (CMake 3.12)](https://cmake.org/cmake/help/v3.12/command/target_link_libraries.html#linking-object-libraries).