roboticslab-uc3m/openrave-yarp-plugins #65: openraveYarpForceEstimator crashing after loading the env

RaulFdzbis, 2018-01-09:
So when trying to execute openraveYarpForceEstimator, the program crash after correctly loading the environment.

```bash
[plugindatabase.h:645 Create] Failed to create name fcl_, interface collisionchecker
[xmlreaders-core.cpp:2599 endElement] failed to create iksolver ik/ikfast0x10000048.Transform6D.x86_64.16_17_18_19_20_21.so
[xmlreaders-core.cpp:2599 endElement] failed to create iksolver ik/ikfast0x10000048.Transform6D.x86_64.22_23_24_25_26_27.so
[info] OpenraveYarpPluginLoader.cpp:112 Open(): Checking for yarp network...
[success] OpenraveYarpPluginLoader.cpp:118 Open(): Found yarp network.
[debug] OpenraveYarpPluginLoader.cpp:142 Open(): config: (robotIndex 0) (manipulatorIndex 2) (genRefSpeed 999999) (device controlboardwrapper2) (subdevice YarpOpenraveControlboard)
[info] OpenraveYarpPluginLoader.cpp:145 Open(): penv: 0x2250d50
[DEBUG]ControlBoardWrapper2: 'period' parameter missing, using default thread period = 20ms 
[DEBUG]opening controlBoardWrapper2 subdevice
[debug] DeviceDriverImpl.cpp:14 open(): config: (penv {80 13 37 2 0 0 0 0 0 20 37 2 0 0 0 0}) (robotIndex 0) (manipulatorIndex 2) (genRefSpeed 999999) (name "/teoSim/rightArm") (device YarpOpenraveControlboard) (subdevice YarpOpenraveControlboard)
[debug] DeviceDriverImpl.cpp:51 open(): Get JointPtr for manipulatorIDs[0]: 4
[debug] DeviceDriverImpl.cpp:51 open(): Get JointPtr for manipulatorIDs[1]: 5
[debug] DeviceDriverImpl.cpp:51 open(): Get JointPtr for manipulatorIDs[2]: 6
[debug] DeviceDriverImpl.cpp:51 open(): Get JointPtr for manipulatorIDs[3]: 7
[debug] DeviceDriverImpl.cpp:51 open(): Get JointPtr for manipulatorIDs[4]: 8
[debug] DeviceDriverImpl.cpp:51 open(): Get JointPtr for manipulatorIDs[5]: 9
[debug] DeviceDriverImpl.cpp:67 open(): pcontrol: 0x2ec2690, IdealController
[info] DeviceDriverImpl.cpp:73 open(): Detected idealcontroller, switch to genericmulticontroller.
[debug] DeviceDriverImpl.cpp:86 open(): pcontrol: 0x42f5fe0, genericmulticontroller
[debug] DeviceDriverImpl.cpp:103 open(): Attach individual controller for manipulatorIDs[0]: 4
[debug] DeviceDriverImpl.cpp:103 open(): Attach individual controller for manipulatorIDs[1]: 5
[debug] DeviceDriverImpl.cpp:103 open(): Attach individual controller for manipulatorIDs[2]: 6
[debug] DeviceDriverImpl.cpp:103 open(): Attach individual controller for manipulatorIDs[3]: 7
[debug] DeviceDriverImpl.cpp:103 open(): Attach individual controller for manipulatorIDs[4]: 8
[debug] DeviceDriverImpl.cpp:103 open(): Attach individual controller for manipulatorIDs[5]: 9
[INFO]created device <YarpOpenraveControlboard>. See C++ class roboticslab::YarpOpenraveControlboard for documentation.
[info] IPositionControlImpl.cpp:9 getAxes(): Reporting 6 axes are present
[DEBUG]joints parameter is 6
[info] IPositionControlImpl.cpp:9 getAxes(): Reporting 6 axes are present
[info] IPositionControlImpl.cpp:9 getAxes(): Reporting 6 axes are present
[INFO]/teoSim/rightArm : no ROS initialization required 
[INFO]/teoSim/rightArm  initting YARP initialization 
yarp: Port /teoSim/rightArm/rpc:i active at tcp://163.117.150.58:10002/
yarp: Port /teoSim/rightArm/command:i active at tcp://163.117.150.58:10003/
yarp: Port /teoSim/rightArm/state:o active at tcp://163.117.150.58:10004/
yarp: Port /teoSim/rightArm/stateExt:o active at tcp://163.117.150.58:10005/
[INFO]created wrapper <controlboardwrapper2>. See C++ class yarp::dev::ControlBoardWrapper2 for documentation.
[success] OpenraveYarpPluginLoader.cpp:297 Open(): Valid yarp plugin.

[info] OpenraveYarpPluginLoader.cpp:112 Open(): Checking for yarp network...
[success] OpenraveYarpPluginLoader.cpp:118 Open(): Found yarp network.
[debug] OpenraveYarpPluginLoader.cpp:142 Open(): config: (robotIndex 0) (manipulatorIndex 0) (genRefSpeed 999999) (device controlboardwrapper2) (subdevice YarpOpenraveControlboard)
[info] OpenraveYarpPluginLoader.cpp:145 Open(): penv: 0x2250d50
[DEBUG]ControlBoardWrapper2: 'period' parameter missing, using default thread period = 20ms 
[DEBUG]opening controlBoardWrapper2 subdevice
[debug] DeviceDriverImpl.cpp:14 open(): config: (penv {80 13 37 2 0 0 0 0 0 20 37 2 0 0 0 0}) (robotIndex 0) (manipulatorIndex 0) (genRefSpeed 999999) (name "/teoSim/trunk") (device YarpOpenraveControlboard) (subdevice YarpOpenraveControlboard)
[debug] DeviceDriverImpl.cpp:51 open(): Get JointPtr for manipulatorIDs[0]: 0
[debug] DeviceDriverImpl.cpp:51 open(): Get JointPtr for manipulatorIDs[1]: 1
[debug] DeviceDriverImpl.cpp:67 open(): pcontrol: 0x42f5fe0, genericmulticontroller
[info] DeviceDriverImpl.cpp:79 open(): Detected genericmulticontroller, which will be used.
[debug] DeviceDriverImpl.cpp:86 open(): pcontrol: 0x42f5fe0, genericmulticontroller
[debug] DeviceDriverImpl.cpp:103 open(): Attach individual controller for manipulatorIDs[0]: 0
[debug] DeviceDriverImpl.cpp:103 open(): Attach individual controller for manipulatorIDs[1]: 1
[INFO]created device <YarpOpenraveControlboard>. See C++ class roboticslab::YarpOpenraveControlboard for documentation.
[info] IPositionControlImpl.cpp:9 getAxes(): Reporting 2 axes are present
[DEBUG]joints parameter is 2
[info] IPositionControlImpl.cpp:9 getAxes(): Reporting 2 axes are present
[info] IPositionControlImpl.cpp:9 getAxes(): Reporting 2 axes are present
[INFO]/teoSim/trunk : no ROS initialization required 
[INFO]/teoSim/trunk  initting YARP initialization 
yarp: Port /teoSim/trunk/rpc:i active at tcp://163.117.150.58:10019/
yarp: Port /teoSim/trunk/command:i active at tcp://163.117.150.58:10020/
yarp: Port /teoSim/trunk/state:o active at tcp://163.117.150.58:10021/
yarp: Port /teoSim/trunk/stateExt:o active at tcp://163.117.150.58:10022/
[INFO]created wrapper <controlboardwrapper2>. See C++ class yarp::dev::ControlBoardWrapper2 for documentation.
[success] OpenraveYarpPluginLoader.cpp:297 Open(): Valid yarp plugin.

[info] OpenraveYarpForceEstimator.cpp:249 Open(): Checking for yarp network...
[success] OpenraveYarpForceEstimator.cpp:255 Open(): Found yarp network.
[debug] OpenraveYarpForceEstimator.cpp:279 Open(): config: (wrinkleSize 4)
[info] OpenraveYarpForceEstimator.cpp:282 Open(): port name: /openraveYarpForceEstimator/rpc:s
[info] OpenraveYarpForceEstimator.cpp:285 Open(): wrinkle Size: 4
[OpenraveYarpForceEstimator.cpp:287 Open] penv: 0x2250d50
sucess: object "object" exists.
sucess: object "ironWrinkle" exists.
Robot 0: teoSim
yarp: Port /openraveYarpForceEstimator/rpc:s active at tcp://163.117.150.58:10023/
None
terminate called after throwing an instance of 'OpenRAVE::openrave_exception'
  what():  openrave (Assert): [/usr/include/boost/smart_ptr/shared_ptr.hpp:653] -> typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = OpenRAVE::KinBody; typename boost::detail::sp_member_access<T>::type = OpenRAVE::KinBody*], expr: px != 0
Aborted (core dumped)
```

jgvictores, 2018-01-09:
Could you point the command/program you are using, and the specific environment?

RaulFdzbis, 2018-01-09:
Yes sure,

I am working in the [develop branch](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/tree/4c95859b5918f329de78ee0c950fbaf6d9ee698b), and the script i am using is this [one](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/4c95859b5918f329de78ee0c950fbaf6d9ee698b/examples/openraveYarpForceEstimator.py).

I am working outside conda and rllab environment so that should not be the problem.

jgvictores, 2018-01-09:
Okay, the bug seems to be because of a confusion regarding the use of these variables: https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/4c95859b5918f329de78ee0c950fbaf6d9ee698b/libraries/OpenravePlugins/OpenraveYarpForceEstimator/OpenraveYarpForceEstimator.cpp#L125-L130

Specifically, this: https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/4c95859b5918f329de78ee0c950fbaf6d9ee698b/libraries/OpenravePlugins/OpenraveYarpForceEstimator/OpenraveYarpForceEstimator.cpp#L283-L291
And this: https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/4c95859b5918f329de78ee0c950fbaf6d9ee698b/libraries/OpenravePlugins/OpenraveYarpForceEstimator/OpenraveYarpForceEstimator.cpp#L145-L146
Should match.

RaulFdzbis, 2018-01-10:
You were right, there were some bugs in the code related to that variables. Thank you @jgvictores!!

After the last two commits the problem is solved. 

I think we can close this now.

closed 2018-01-10 by RaulFdzbis