roboticslab-uc3m/openrave-yarp-plugins #66: OpenraveYarpForceEstimator improve and cleanup

jgvictores, 2018-01-10:
As a followup of #65, still some improvements to do in `OpenraveYarpForceEstimator `.
- Clean unused vars similar to https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/develop/libraries/OpenravePlugins/OpenraveYarpForceEstimator/OpenraveYarpForceEstimator.cpp#L124-L130
- Review chunks of commented code, such as: https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/develop/libraries/OpenravePlugins/OpenraveYarpForceEstimator/OpenraveYarpForceEstimator.cpp#L322-L359
- See if there is a grab return value that can be checked: https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/develop/libraries/OpenravePlugins/OpenraveYarpForceEstimator/OpenraveYarpForceEstimator.cpp#L309

RaulFdzbis, 2018-01-10:
Agree with that! and I can help if wanted/needed :D

jgvictores, 2018-01-10:
Ok, no hurry! Just unassigned myself, and put `priority: low` label. :-)

jgvictores, 2018-01-11:
Some WIP at 82118e0971f4b6e56361e4ea070a026538c55a88

jgvictores, 2018-01-11:
More WIP at 82af6b386681ea0ad6c990ddbd80188bdba5d0fd

jgvictores, 2018-01-12:
More WIP at 08ae8049b4d7c42fa721a348b5cb16aae558a748. No more WIP for now.

jgvictores, 2018-01-14:
Blocks #68