roboticslab-uc3m/openrave-yarp-plugins #67: Output error when calling openraveYarpForceEstimator from xgnitive

RaulFdzbis, 2018-01-10:
So when calling openraveYarpForceEstimator using this [program](https://github.com/roboticslab-uc3m/xgnitive/blob/7df1a81d04031d5caf2cca3739438e9968e20f87/programs/drl/envs/teo_iron_discrete_yarp_env.py#L57-L59).  I get this error:

>[error] YarpOpenraveBase.cpp:171 configureOpenravePlugins(): Bad send 'open --name /drl/openraveYarpForceEstimator/rpc:s --wrinkleSize 4' command.

Followed by:

>[success] YarpOpenraveBase.cpp:173 configureOpenravePlugins(): Sent 'open --name /drl/openraveYarpForceEstimator/rpc:s --wrinkleSize 4' command.

The program works normally. So i dont know if this is an expected behaviour.  I am working with the develop branch of this repo, and the program is executed inside the rllab environment. The full output of the program is the following:

```bash
[debug] DeviceDriverImpl.cpp:14 open(): config: (orPlugins (OpenraveYarpForceEstimator (module OpenraveYarpForceEstimator) (commands "open --name /drl/openraveYarpForceEstimator/rpc:s --wrinkleSize 4"))) (env "/usr/local/share/xgnitive/contexts/models/teo_cgda_iros_iron.env.xml") (robotIndex 0) (manipulatorIndex 2) (genRefSpeed 9999999) (name "/drl/rightArm") (device YarpOpenraveControlboard) (subdevice YarpOpenraveControlboard)
[debug] YarpOpenraveBase.cpp:45 configureEnvironment(): Found --env parameter.
[plugindatabase.h:645 Create] Failed to create name fcl_, interface collisionchecker
[xmlreaders-core.cpp:2599 endElement] failed to create iksolver ik/ikfast0x10000048.Transform6D.x86_64.16_17_18_19_20_21.so
[xmlreaders-core.cpp:2599 endElement] failed to create iksolver ik/ikfast0x10000048.Transform6D.x86_64.22_23_24_25_26_27.so
[success] YarpOpenraveBase.cpp:66 configureEnvironment(): Loaded environment '/usr/local/share/xgnitive/contexts/models/teo_cgda_iros_iron.env.xml'.
[debug] YarpOpenraveBase.cpp:126 configureOpenravePlugins(): Found --orPlugins parameter.
[debug] YarpOpenraveBase.cpp:134 configureOpenravePlugins(): orPlugins: orPlugins (OpenraveYarpForceEstimator (module OpenraveYarpForceEstimator) (commands "open --name /drl/openraveYarpForceEstimator/rpc:s --wrinkleSize 4"))
[debug] YarpOpenraveBase.cpp:143 configureOpenravePlugins(): orPlugin[1]: OpenraveYarpForceEstimator (module OpenraveYarpForceEstimator) (commands "open --name /drl/openraveYarpForceEstimator/rpc:s --wrinkleSize 4")
[debug] YarpOpenraveBase.cpp:145 configureOpenravePlugins(): * orPlugin[1]: plugin: OpenraveYarpForceEstimator
[debug] YarpOpenraveBase.cpp:147 configureOpenravePlugins(): * orPlugin[1]: module: OpenraveYarpForceEstimator
[debug] YarpOpenraveBase.cpp:150 configureOpenravePlugins(): * orPlugin[1]: commands: open --name /drl/openraveYarpForceEstimator/rpc:s --wrinkleSize 4
[info] OpenraveYarpForceEstimator.cpp:276 Open(): Checking for yarp network...
[success] OpenraveYarpForceEstimator.cpp:282 Open(): Found yarp network.
[debug] OpenraveYarpForceEstimator.cpp:306 Open(): config: (wrinkleSize 4) (name "/drl/openraveYarpForceEstimator/rpc:s")
[info] OpenraveYarpForceEstimator.cpp:309 Open(): port name: /drl/openraveYarpForceEstimator/rpc:s
[info] OpenraveYarpForceEstimator.cpp:312 Open(): wrinkle Size: 4
[OpenraveYarpForceEstimator.cpp:314 Open] penv: 0x35aedd0
sucess: object "object" exists.
sucess: object "ironWrinkle" exists.
Robot 0: teoSim
yarp: Port /drl/openraveYarpForceEstimator/rpc:s active at tcp://163.117.150.58:10019/
[error] YarpOpenraveBase.cpp:171 configureOpenravePlugins(): Bad send 'open --name /drl/openraveYarpForceEstimator/rpc:s --wrinkleSize 4' command.
[success] YarpOpenraveBase.cpp:173 configureOpenravePlugins(): Sent 'open --name /drl/openraveYarpForceEstimator/rpc:s --wrinkleSize 4' command.
[debug] DeviceDriverImpl.cpp:51 open(): Get JointPtr for manipulatorIDs[0]: 4
[debug] DeviceDriverImpl.cpp:51 open(): Get JointPtr for manipulatorIDs[1]: 5
[debug] DeviceDriverImpl.cpp:51 open(): Get JointPtr for manipulatorIDs[2]: 6
[debug] DeviceDriverImpl.cpp:51 open(): Get JointPtr for manipulatorIDs[3]: 7
[debug] DeviceDriverImpl.cpp:51 open(): Get JointPtr for manipulatorIDs[4]: 8
[debug] DeviceDriverImpl.cpp:51 open(): Get JointPtr for manipulatorIDs[5]: 9
[debug] DeviceDriverImpl.cpp:67 open(): pcontrol: 0x409e790, IdealController
[info] DeviceDriverImpl.cpp:73 open(): Detected idealcontroller, switch to genericmulticontroller.
[debug] DeviceDriverImpl.cpp:86 open(): pcontrol: 0x50f7530, genericmulticontroller
[debug] DeviceDriverImpl.cpp:103 open(): Attach individual controller for manipulatorIDs[0]: 4
[debug] DeviceDriverImpl.cpp:103 open(): Attach individual controller for manipulatorIDs[1]: 5
[debug] DeviceDriverImpl.cpp:103 open(): Attach individual controller for manipulatorIDs[2]: 6
[debug] DeviceDriverImpl.cpp:103 open(): Attach individual controller for manipulatorIDs[3]: 7
[debug] DeviceDriverImpl.cpp:103 open(): Attach individual controller for manipulatorIDs[4]: 8
[debug] DeviceDriverImpl.cpp:103 open(): Attach individual controller for manipulatorIDs[5]: 9
[INFO]created device <YarpOpenraveControlboard>. See C++ class roboticslab::YarpOpenraveControlboard for documentation.
[info] IPositionControlImpl.cpp:9 getAxes(): Reporting 6 axes are present
[DEBUG]joints parameter is 6
[info] IPositionControlImpl.cpp:9 getAxes(): Reporting 6 axes are present
[info] IPositionControlImpl.cpp:9 getAxes(): Reporting 6 axes are present
[INFO]/drl/rightArm : no ROS initialization required 
[INFO]/drl/rightArm  initting YARP initialization 
yarp: Port /drl/rightArm/rpc:i active at tcp://163.117.150.58:10017/
yarp: Port /drl/rightArm/command:i active at tcp://163.117.150.58:10018/
yarp: Port /drl/rightArm/state:o active at tcp://163.117.150.58:10021/
yarp: Port /drl/rightArm/stateExt:o active at tcp://163.117.150.58:10022/
[INFO]created wrapper <controlboardwrapper2>. See C++ class yarp::dev::ControlBoardWrapper2 for documentation.
yarp: Port /cgda/openraveYarpForceEstimator/rpc:c active at tcp://163.117.150.58:10026/
yarp: Sending output from /cgda/openraveYarpForceEstimator/rpc:c to /drl/openraveYarpForceEstimator/rpc:s using tcp
Wait to connect to force server...

yarp: Receiving input from /cgda/openraveYarpForceEstimator/rpc:c to /drl/openraveYarpForceEstimator/rpc:s using tcp
```

jgvictores, 2018-01-10:
Okay, so you just discovered a bug in `YarpOpenraveBase`. There should be a `return false;` after:
https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/82af6b386681ea0ad6c990ddbd80188bdba5d0fd/libraries/YarpPlugins/YarpOpenraveBase/YarpOpenraveBase.cpp#L119

Now, investigation why we are getting an error, it seems the code is outdated and we are only sending and `open`, not the complete command:
https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/82af6b386681ea0ad6c990ddbd80188bdba5d0fd/libraries/YarpPlugins/YarpOpenraveBase/YarpOpenraveBase.cpp#L114-L117

This is, `name` etc are not actually getting through, you're just getting the defaults. Let me start some fixes on a branch.

jgvictores, 2018-01-11:
Added missing `return false;` on `YarpOpenraveBase` at 2fe53a88fcd3dba4f5dbcc469ddf2e8a87bfb4c1.
Todo:
1. Recover `return true;` on `openraveYarpForceEstimator`.
1. Pass all `open` parameters at `YarpOpenraveBase`.

jgvictores, 2018-01-12:
Should be fixed with 08ae8049b4d7c42fa721a348b5cb16aae558a748 and 2fe53a88fcd3dba4f5dbcc469ddf2e8a87bfb4c1.

> Pass all open parameters at YarpOpenraveBase.

Starting new issue on this.

closed 2018-01-12 by jgvictores