roboticslab-uc3m/openrave-yarp-plugins #68: Introduce orientation requirment to iron.

RaulFdzbis, 2018-01-11:
TODO, introduce orientation requirment in iron. Options:

- Using quaternions angles (default way to work with orientation in openrave).
- Using euler angles (convert quaternions to euler angles) less efficent but conceptually simpler.

I am working in this.

jgvictores, 2018-01-11:
WIP at a5ede71964121cf194e56dbd44997dcb498b7508

jgvictores, 2018-01-11:
More WIP at 083c5556de6e790b35c07ab728fe036fcd0897db

RaulFdzbis, 2018-01-12:
After new priority organization, the development of this issue will be stopped for the moment.

David-Estevez, 2018-01-12:
I've been speaking to @RaulFdzbis this morning and he told me to write here this idea:

What about not only having a positive reward for ironing a region with a certain orientation, but also to give a negative reward if the robot tries to iron a region with an incorrect region? This region could be even an already ironed one, that becomes wrinkled again if ironed incorrectly.

jgvictores, 2018-01-12:
Deleted the branch `issue-68-ori` which currently simply pointed to existing 2fe53a88fcd3dba4f5dbcc469ddf2e8a87bfb4c1.

jgvictores, 2018-01-14:
Blocked by #66