roboticslab-uc3m/openrave-yarp-plugins #76: Failed to find command 'setvelocity' in interface idealcontroller

PeterBowman, 2018-02-05:
Current develop (bcf174e), [asibotSim](https://github.com/roboticslab-uc3m/asibot-configuration-files/blob/406c38bd7ee42f2c31815305994deddda14087d4/scripts/bash/asibotSim) script. Steps:

* `set poss (0 -30 90 30 0)` via RPC (simulation)
* `movl 0.55 0 0.95 0 1.57` via RPC (BasicCartesianControl + KdlSolver)

Output and crash message:

```
[debug] IControlMode2Impl.cpp:84 setControlMode(): (0, 7103862)
[debug] IControlMode2Impl.cpp:84 setControlMode(): (1, 7103862)
[debug] IControlMode2Impl.cpp:84 setControlMode(): (2, 7103862)
[debug] IControlMode2Impl.cpp:84 setControlMode(): (3, 7103862)
[debug] IControlMode2Impl.cpp:84 setControlMode(): (4, 7103862)
[info] IVelocityControl2Impl.cpp:9 velocityMove(): 
terminate called after throwing an instance of 'OpenRAVE::openrave_exception'
  what():  openrave (CommandNotSupported): failed to find command 'setvelocity' in interface idealcontroller

Aborted (core dumped)
```

PeterBowman, 2018-02-05:
Unless I'm wrong, these are the only commands that [IdealController](http://openrave.org/docs/0.8.0/interface_types/controller/idealcontroller/) accepts: [ref](https://github.com/rdiankov/openrave/blob/b2818fb534cd8c4e64ea1dddea0f7c360ce5056d/plugins/basecontrollers/idealcontroller.cpp#L32-L39). What about [IdealVelocityController](http://openrave.org/docs/0.8.2/interface_types/controller/idealvelocitycontroller/)?

jgvictores, 2018-02-05:
I have a long discussion about controllers on the related `epic` issue.
From what I recall:
1. Use setDesired
1. Must activate ODE :-(

PeterBowman, 2018-02-05:
> I have a long discussion about controllers on the related epic issue.

#29

jgvictores, 2018-02-05:
I'm pretty sure `setvelocity` was for `odevelocity`, which as commented [here](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/29#issuecomment-340159347) is obsolete.
For `idealvelocitycontroller`, I'm pretty sure the `setDesired()` method would work.
However, since `idealvelocitycontroller` requires ODE, my plan was to keep using the position controller, but send to joint limits. The basic method is [this one](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/develop/libraries/YarpPlugins/YarpOpenraveControlboard/IPositionControlImpl.cpp#L15), but must take into account that openrave trajectories are set in duration not velocity (careful with infinite or max joint limits here).

jgvictores, 2018-02-05:
...which additionally translates to: we unfortunately do not have a functional `velocityMove` (or basically anything of `IVelocityControl`) in `YarpOpenraveControlboard` right now... :-(

PeterBowman, 2018-02-09:
Raising priority. Sadly, most control is performed in velocity mode. 😞

jgvictores, 2018-02-09:
Can be fixed today. Just have to speak and decide between the 2-3 options.

jgvictores, 2018-02-09:
Mostly implemented at 5c1e03c215e9e8fc7306f4ec7858455fcd5f3144.
Small fixes until current `develop`, at 7a7c2e9cbabda5c787abfdc2b1d6f3c76989498a.

Good enough for `movl` on AMOR!

closed 2018-02-09 by jgvictores

jgvictores, 2018-02-10:
Tested and working also with "no" joint limits, via `<Joint circular="false" .........>` (where in fact joint limit values are just set to very large values). :tada: