roboticslab-uc3m/openrave-yarp-plugins #77: How to colour a wall

RaulFdzbis, 2018-02-08:
_From @RaulFdzbis on January 18, 2018 11:28_

What problems/step we need to solve in order to create a paint action that allows the robot to paint the wall in different colours:

- [x] Modify openraveYarpPaintSquares plugin to paint the wall in different colours as a function of a variable ("brushColour"). I think the best idea is to define this variable inside this plugin.
- [x] Allow the robot to modify the "brushColour" variable by place the EEF in some point ("palete") of the working space.
- [x] Add some visual element to define the "palete" in the working space. Better debugging and presentation.

_Copied from original issue: RaulFdzbis/raul-research-shop#7_

RaulFdzbis, 2018-02-08:
_From @jgvictores on January 18, 2018 14:5_

Yes, these are the steps. I think they can all be incorporated in `openraveYarpPaintSquares`. Please update https://github.com/roboticslab-uc3m/openrave-yarp-plugins and start a branch off develop for this!

RaulFdzbis, 2018-02-08:
Yeah i have already done that :D https://github.com/roboticslab-uc3m/openrave-yarp-plugins/tree/wall-paint-colour-plugin

RaulFdzbis, 2018-02-08:
_From @jgvictores on January 18, 2018 15:55_

> Yeah i have already done that :D https://github.com/roboticslab-uc3m/openrave-yarp-plugins/tree/wall-paint-colour-plugin

This is good! Thanks!

RaulFdzbis, 2018-02-08:
with the last commit https://github.com/roboticslab-uc3m/openrave-yarp-plugins/commit/373c2a1e37f95982b637a3c4ca2840a5ea61d1a8 this issue is almost done. Now we only have to check that the implementation works, and also find a nice place with a good reachability for the palete in the work space.

[image]

RaulFdzbis, 2018-02-08:
_From @jgvictores on January 20, 2018 11:48_

Looks fantastic! Do you think there is any possibility of putting the changer somewhere where the other hand can reach, to make it look more like a classical artist wooden palette?

RaulFdzbis, 2018-02-08:
The thing is reachability is a problem here. The dimensions of the wall are chosen taking care of this. So placing the palette to the left of the wall, would be too difficult for the robot to reach. I think the best idea is placing it below the wall, just as paint bottles.

Btw, i was talking with Olaya and told me that the basic colours for painting are yellow, green, blue, so maybe is a better idea to change to that colours.

RaulFdzbis, 2018-02-08:
_From @jgvictores on January 22, 2018 11:10_

> below the wall, just as paint bottles

Could make sense, let's give it a try.

> Btw, i was talking with Olaya and told me that the basic colours for painting are yellow, green, blue, so maybe is a better idea to change to that colours.

Cool, let's do it!

RaulFdzbis, 2018-02-08:
Its working!

[image]

The only pending thing to do is to change the colours to yellow, red, blue then

RaulFdzbis, 2018-02-08:
Now using painting basic colours (ﾉ◕ヮ◕)ﾉ*:･ﾟ✧
 
[image]

RaulFdzbis, 2018-02-08:
This task, can now considered to be done. Some future steps would be to allow mixing of colours. In simulation, is easy to do. An idea is simply sum the value of the rgb colour of the brush plus the rgb colour of the square, and assign that value to the square.

Right now this is out of the scope of the action we are designing.

RaulFdzbis, 2018-02-08:
_From @jgvictores on January 23, 2018 10:45_

> Right now this is out of the scope of the action we are designing.

...or not? :wink:

jgvictores, 2018-03-22:
Done at around d5c7c0dbcbac57304df8caa1941d5d4bbb363474.

closed 2018-03-22 by jgvictores