roboticslab-uc3m/openrave-yarp-plugins #83: Collision model (convex decomposition) must be re-generated each run

jgvictores, 2018-03-14:
Collision model must be re-generated each run, namely [these lines](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/94576dc4d57d5a13d70de239ec01510442d5fee0/examples/python/openraveYarpPluginLoader-controlboard-collision.py#L19-L37).

Possible hint at runtime:
```bash
No handlers could be found for logger "openravepy.databases.convexdecomposition"
```

RaulFdzbis, 2018-03-14:
This is an already known behaviour with convexdecomposition, the problem is that when the convex decomposition model (cdmdel) is generated, this cdmodel is saved in the openrave database as a function of it's geometry. If some padding is done in the cdmodel, the geometry changes and therefore its hash. **Since the "load" function always tries to load the not padded cdmodel**, this will result in the program generating a new convexdecomposition model each time. 

This behaviour has also been reported in the openrave [forum](http://openrave-users-list.185357.n3.nabble.com/Re-Hand-Jacobian-td3506786.html).

Until its not fixed, a possible solution is to manually rename the padded model (robot_padding.pp) to have the same name as the not padded one (robot.pp).

jgvictores, 2018-03-14:
Or at least slow... Some pointers:
- #8 
- Launching ([ref](http://openrave.org/docs/latest_stable/openravepy/databases.convexdecomposition/#usage))
   ```bash
   openrave.py --database convexdecomposition --robot=/usr/local/share/teo-openrave-models/contexts/openrave/teo/teo.robot.xml --padding=0.02 --show --getfilename
   ```
   Returns:
   ```
   /home/yo/.openrave/robot.29d30d6ed41a0d1254635867fcebcd97/convexdecomposition_0.020.pp
   ```
- Files in commit https://github.com/roboticslab-uc3m/teo-openrave-models/commit/984a27d72b7f547239c59800c479bc588799f599

jgvictores, 2018-03-14:
More refs, freshly moved into this repo: #84

jgvictores, 2018-03-14:
From what was commented at https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/83#issuecomment-372999608 and with @RaulFdzbis in person, should be a very simple issue related to path names. Possibly due to https://github.com/rdiankov/openrave/issues/118, OpenRAVE does not look for the padded version.

jgvictores, 2018-03-14:
> No handlers could be found for logger "openravepy.databases.convexdecomposition"

For the record, this happens upon [cdmodel.load()](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/94576dc4d57d5a13d70de239ec01510442d5fee0/examples/python/openraveYarpPluginLoader-controlboard-collision.py#L24) returning False.

jgvictores, 2018-03-14:
For the record, `cdmodel.generate(padding=0.02)` takes about 2 minutes, and uses 2 of a 4 core machine (1 constant, switching the other).

jgvictores, 2018-03-14:
`cdmodel.save()` takes 5 s, generates the 60 MB model `robot.29d30d6ed41a0d1254635867fcebcd97/convexdecomposition_0.020.pp` (note the same hash, this is good).

jgvictores, 2018-03-14:
`cdmodel.setrobot()` takes another 10-15 s.

jgvictores, 2018-03-14:
`cdmodel.show()` possibly stuck in [while True](https://github.com/rdiankov/openrave/blob/7c5f5e27eec2b2ef10aa63fbc519a998c276f908/python/databases/convexdecomposition.py#L560).

jgvictores, 2018-03-14:
...but can be interrupted in a python console via ctrl-c. :-)
```python
^CTraceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_/databases/convexdecomposition.py", line 563, in show
    time.sleep(0.5)
KeyboardInterrupt
```

jgvictores, 2018-03-14:
Fixed at 756a3a3d57bc4e781e6dc13814ace3abca44d776. The most important change was to specify the padding in the constructor, before load(), as [here](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/756a3a3d57bc4e781e6dc13814ace3abca44d776/examples/python/openraveYarpPluginLoader-controlboard-collision.py#L21). Minimal working example (see how paddings must match):
```python
import openravepy
from openravepy import *
RaveInitialize()

env=Environment()
env.SetViewer('qtcoin')
env.Load('/usr/local/share/teo-openrave-models/contexts/openrave/teo/teo.robot.xml')

# Convex Decomposition
teo_robot = env.GetRobots()[0]
cdmodel = databases.convexdecomposition.ConvexDecompositionModel(teo_robot, padding=0.02)  # Else defaults to zero padding

if not cdmodel.load(): 
    # If not already in the database. Generate:
    print 'ConvexDecomposition not found, generating...'
    cdmodel.generate(padding=0.02)  # Else defaults to 0.005 padding
    print 'ConvexDecomposition generated, saving...'
    cdmodel.save()
    print 'Finished saving'

print 'Setting robot...'
cdmodel.setrobot()
print 'Finish setrobot'

print 'Starting visualization of the convexdecomposition model'
cdmodel.show()   # blocking for now
```

Note how `padding` must be the same (and defaults are different).

jgvictores, 2018-03-14:
Times are reduced from 2:25 to 0:25.

jgvictores, 2018-03-14:
Merged into `develop` at 94a8b29b7ab38dfd469ee0567ee1640fec3143fe.

closed 2018-03-14 by jgvictores

jgvictores, 2018-04-04:
Working script for TEO model at https://github.com/roboticslab-uc3m/teo-openrave-models/blob/6cd40916d8637043d1c2ff587f6322a8cab3ffdf/scripts/python/generateConvexDecomposition.py