roboticslab-uc3m/openrave-yarp-plugins #86: gripperjoints

jgvictores, 2018-04-04:
For https://github.com/roboticslab-uc3m/teo-openrave-models/issues/9 it seems like we should do something with joints/gripperjoints:

https://github.com/rdiankov/openrave/blob/7c5f5e27eec2b2ef10aa63fbc519a998c276f908/src/libopenrave-core/xmlreaders-core.cpp#L2516 we have:
`(xmlname == "joints")||(xmlname == "gripperjoints")`

An alternative hack is to just keep on using manipulator.

jgvictores, 2018-04-04:
The test platform will be simulated teo, see XML:
https://github.com/roboticslab-uc3m/teo-openrave-models/blob/0efd49e662495f8ca329ac0089fc37243ab7519c/openrave/teo/teo.robot.xml#L76

jgvictores, 2019-11-20:
Towards https://github.com/roboticslab-uc3m/teo-main/issues/19:
1. It turns out `<gripperjoints>` is not required for what we need As of https://github.com/roboticslab-uc3m/teo-openrave-models/commit/82163a8a6c2fbbc3fae807f253eeaef58030efbf plus new issue https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/97 are the way to go.
2. Added ref to this issue in "Think about openrave-database-generators" (https://github.com/roboticslab-uc3m/teo-openrave-models/issues/3) in case ever needed.

Closing due to (1).

PeterBowman, 2019-11-20:
What is `<gripperjoints>` meant for? Could we ever need it for some different application?

jgvictores, 2019-11-20:
From what I've seen, it's used for OpenRAVE's database generators, e.g. inverse kinematics and grasping. This is why I referenced this issue at https://github.com/roboticslab-uc3m/teo-openrave-models/issues/3 (Think about openrave-database-generators).

jgvictores, 2019-11-21:
Reopening, sorry. Not bad to have a specific issue to recover this.

jgvictores, 2020-01-23:
> * It turns out `<gripperjoints>` is not required for what we need As of [roboticslab-uc3m/teo-openrave-models@82163a8](https://github.com/roboticslab-uc3m/teo-openrave-models/commit/82163a8a6c2fbbc3fae807f253eeaef58030efbf) plus new issue https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/97 are the way to go.

I confirm the `<joints>` tag (aka `<gripperjoints>` afaik) is required for the OpenRAVE grasping module (`databases.grasping.GraspingModel(robot, target)`). Recovered for TEO at https://github.com/roboticslab-uc3m/teo-openrave-models/commit/083998bfde1d05dcd96b7626da8a8c3bc8edf719

closed 2020-01-23 by jgvictores