roboticslab-uc3m/openrave-yarp-plugins #87: YarpOpenraveRGBDSensor publishes fixed positions of obstacles

PeterBowman, 2018-05-28:
Despite correctly sending information from available depth sensors through the `<prefix>/depthImage:o` port, all coordinates refer to the absolute (fixed) location of obstacles along one of the environment's axes. That is, it doesn't matter where the sensor is located at. It's worth noting that a quick glimpse at yarpview's output will not reveal any of this, the trouble starts when working with the actual data.

To reproduce it, launch [this example](https://github.com/jgvictores/2018-ptmr/blob/1f708efa7ea4dca1a9fd6bb0f898f35ea4a9763f/map1/openrave/ptmrSim1) and make the robot face a *wall*, looking in the direction of the *y* axis. Pixel values won't change if the sensor is translated along said axis. To make this even more obvious, focus the same wall from an oblique angle (values won't change either if belonging to the same plane), or try with [slopes](https://github.com/jgvictores/2018-ptmr/tree/1f708efa7ea4dca1a9fd6bb0f898f35ea4a9763f/map7).

While debugging, I learned that sensor data (`sensorRanges`) resolves to a matrix of zeroes, which leads me to think that it's more an OpenRAVE than a YARP issue.

https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/e27f1cfb74b7f98ca805ccdd1663c6fd9a14ed6e/libraries/YarpPlugins/YarpOpenraveRGBDSensor/IRGBDSensorImpl.cpp#L72

jgvictores, 2018-05-28:
Did a quick try via `python openraveYarpPluginLoader-rgbdsensor-noRGB.py `. Ports are open, but get error:
```
[INFO]created device <YarpOpenraveRGBDSensor>. See C++ class roboticslab::YarpOpenraveRGBDSensor for documentation.
[ERROR]RGBD wrapper: error configuring interfaces for parsers 
[ERROR]RGBDSensorWrapper: error while opening subdevice
yarpdev: ***ERROR*** driver <RGBDSensorWrapper> was found but could not open
[info] DeviceDriverImpl.cpp:134 close(): 
[error] OpenraveYarpPluginLoader.cpp:241 Open(): yarp plugin not valid.
None
```
Anything similar on your side?

jgvictores, 2018-05-28:
YARP 5bc7bfb7958b77a8655e390e05b82e993869cb92, OpenRAVE 7c5f5e27eec2b2ef10aa63fbc519a998c276f908 (current master), o-y-p e27f1cfb74b7f98ca805ccdd1663c6fd9a14ed6e (current develop).

Can you replicate the following screenshots?
[image]
[image]
[image]

PeterBowman, 2018-05-28:
Looks pretty much the same.

[image]
[image]
[image]
[image]
[image]
[image]

jgvictores, 2018-05-30:
Note to self: https://github.com/personalrobotics/offscreen_render

PeterBowman, 2018-05-30:
> While debugging, I learned that sensor data (sensorRanges) resolves to a matrix of zeroes

No longer observing this behavior, but depth is still wrong.

jgvictores, 2018-05-30:
Much better thanks to #88 !!
[image]

PR #88 (2018-05-30): Publish actual depth data
`LaserSensorData::ranges` is everything we need: distance to obstacles measured at the sensor's focal point, expressed in *global* axes. Just find the euclidean distance and everything is set. No need to use the transform (from global system to sensor w/ orientation) nor the `LaserSensorData::positions` vector (translation part of said transform matrix).

TL;DR: see [docs](http://openrave.org/docs/latest_stable/coreapihtml/classOpenRAVE_1_1SensorBase_1_1LaserSensorData.html).

Closes #87.
closed 2018-05-30 by jgvictores