roboticslab-uc3m/openrave-yarp-plugins #90: [YarpOpenraveControlboard] joint limit treatment

jgvictores, 2018-07-06:
[YarpOpenraveControlboard] joint limit treatment, as commented f2f @rsantos88 @PeterBowman

PeterBowman, 2018-07-06:
Spotted while working with amorSim: start from initial position (gripper in extended pose), try to pan in -x, first rotational joint affects the orientation of the TCP (with lots of OR warnings).

jgvictores, 2020-01-05:
Note: Warnings are slightly less cryptic applying patches documented at http://robots.uc3m.es/gitbook-installation-guides/install-openrave.html#source-code-hacks ([perma](https://github.com/roboticslab-uc3m/installation-guides/blob/c79885220856a78ce6c0bc2381b59d6485bf9cbe/install-openrave.md#source-code-hacks)).

jgvictores, 2020-02-07:
This turns out to be similar to https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/82, where an issue is we can activate exceptions as in https://github.com/roboticslab-uc3m/openrave-yarp-plugins/commit/bf9e96200c662adbf65b0975b83147b93bf6c801, but would need to implement out own stepSimulation to be able to catch and act.

Note that in the case of joint limits, while warning (or error with exceptions) messages seem out of range, in the visualization we can see the robot stops and, moreover, encoders report having stopped too.