roboticslab-uc3m/openrave-yarp-plugins #92: RGBD sensor plugin must implement yarp::dev::IFrameGrabberControls(2)

PeterBowman, 2018-09-24:
Since YARP 2.3.72.1, [`yarp::dev::RGBDSensorWrapper`](http://www.yarp.it/classyarp_1_1dev_1_1RGBDSensorWrapper.html) expects that its wrapped subdevices implement `yarp::dev::FrameGrabberControls2` ([`yarp::dev::FrameGrabberControls`](http://www.yarp.it/classyarp_1_1dev_1_1IFrameGrabberControls.html) since YARP 3.0).

PeterBowman, 2018-09-25:
See https://github.com/robotology/yarp/issues/1875.

PeterBowman, 2018-09-26:
The aforementioned interface will be made optional at YARP 3.1.1 thanks to https://github.com/robotology/yarp/pull/1878. Anyway, our plugin will remain broken for 2.3.72.1, 3.0.0, 3.0.1 and 3.1.0.

PeterBowman, 2019-04-18:
YARP v3.1.1 was released today.

closed 2019-04-21 by PeterBowman

jgvictores, 2019-04-22:
@PeterBowman Thanks!!

jgvictores, 2020-01-03:
https://www.frontiersin.org/articles/10.3389/frobt.2018.00040/full