roboticslab-uc3m/openrave-yarp-plugins #93: [openrave-yarp-plugins-install.md] Document hard vs soft dependencies

jgvictores, 2018-10-19:
Document hard vs soft dependencies in https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/develop/doc/openrave-yarp-plugins-install.md which BTW seems very outdated (see `ROBOTICSLAB_OPENRAVE_YARP_PLUGINS_DIR`)

Help: See `asrob/yarp-devices` is soft in https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/e51ca867f959060f9859e1bc87661108d887cb4a/CMakeLists.txt#L44-L56

PeterBowman, 2018-10-19:
The [InstallOpenravePlugin.cmake](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/e51ca867f959060f9859e1bc87661108d887cb4a/cmake/modules/InstallOpenravePlugin.cmake) CMake module is the only thing exported by this repo, there are no public headers nor libraries. Since I was considering not to export this anymore, the `ROBOTICSLAB_OPENRAVE_YARP_PLUGINS_DIR` thing would no longer make sense - just in this repo. Perhaps we could start a org-wide discussion on whether such info is actually useful to end-users.

PeterBowman, 2019-06-13:
Actually, I think none of our repos documents soft dependencies, therefore this issue could be treated horizontally.

jgvictores, 2020-01-27:
Closing via https://github.com/roboticslab-uc3m/openrave-yarp-plugins/commit/f612c3e16253cceab18aa6bdff6305cc50023e5a.

Can serve as a precedent for https://github.com/roboticslab-uc3m/questions-and-answers/issues/14

closed 2020-01-27 by jgvictores