roboticslab-uc3m/openrave-yarp-plugins #112: [openraveYarpPaintSquares] Loading sensors error on Bionic

RaulFdzbis, 2020-01-10:
Getting these errors when running python openraveYarpPaintSquares.py on a computer with Bionic and Openrave 0.9 (installed using https://github.com/cielavenir/openrave/tree/boost-1.6x-forcompile) :
```bash
2020-01-10 16:40:31,882 openrave [WARN] [plugindatabase.h:645 InterfaceBasePtr OpenRAVE::RaveDatabase::Create] Failed to create name viewerrecorder, interface module
2020-01-10 16:40:32,635 openrave [WARN] [plugindatabase.h:645 InterfaceBasePtr OpenRAVE::RaveDatabase::Create] Failed to create name forcesensor, interface sensor
2020-01-10 16:40:32,635 openrave [ERROR] [xmlreaders-core.cpp:1805 InterfaceXMLReader::InterfaceXMLReader] xml readers failed to create instance of type sensor:forcesensor
2020-01-10 16:40:32,635 openrave [WARN] [plugindatabase.h:560 InterfaceBasePtr OpenRAVE::RaveDatabase::Create] interface sensor name "" needs to start with a valid character
2020-01-10 16:40:32,635 openrave [ERROR] [xmlreaders-core.cpp:1843 InterfaceXMLReader::_CheckInterface] xml readers failed to create instance of type sensors
2020-01-10 16:40:32,635 openrave [WARN] [plugindatabase.h:560 InterfaceBasePtr OpenRAVE::RaveDatabase::Create] interface sensor name "" needs to start with a valid character
2020-01-10 16:40:32,635 openrave [ERROR] [xmlreaders-core.cpp:1843 InterfaceXMLReader::_CheckInterface] xml readers failed to create instance of type sensors
2020-01-10 16:40:32,635 openrave [WARN] [plugindatabase.h:645 InterfaceBasePtr OpenRAVE::RaveDatabase::Create] Failed to create name forcesensor, interface sensor
2020-01-10 16:40:32,635 openrave [ERROR] [xmlreaders-core.cpp:1805 InterfaceXMLReader::InterfaceXMLReader] xml readers failed to create instance of type sensor:forcesensor
2020-01-10 16:40:32,635 openrave [WARN] [plugindatabase.h:560 InterfaceBasePtr OpenRAVE::RaveDatabase::Create] interface sensor name "" needs to start with a valid character
2020-01-10 16:40:32,635 openrave [ERROR] [xmlreaders-core.cpp:1843 InterfaceXMLReader::_CheckInterface] xml readers failed to create instance of type sensors
2020-01-10 16:40:32,635 openrave [WARN] [plugindatabase.h:560 InterfaceBasePtr OpenRAVE::RaveDatabase::Create] interface sensor name "" needs to start with a valid character
2020-01-10 16:40:32,635 openrave [ERROR] [xmlreaders-core.cpp:1843 InterfaceXMLReader::_CheckInterface] xml readers failed to create instance of type sensors
2020-01-10 16:40:32,912 openrave [WARN] [fclspace.h:248 FCLSpace::KinBodyInfoPtr fclrave::FCLSpace::InitKinBody] Initializing link teoSim/FrontalRightWrist with 0 geometries (env 1) (userdatakey fclcollision0x55ec3464df60)
2020-01-10 16:40:32,912 openrave [WARN] [fclspace.h:248 FCLSpace::KinBodyInfoPtr fclrave::FCLSpace::InitKinBody] Initializing link teoSim/RightArmTCP with 0 geometries (env 1) (userdatakey fclcollision0x55ec3464df60)
2020-01-10 16:40:32,929 openrave [WARN] [fclspace.h:248 FCLSpace::KinBodyInfoPtr fclrave::FCLSpace::InitKinBody] Initializing link teoSim/FrontalLeftWrist with 0 geometries (env 1) (userdatakey fclcollision0x55ec3464df60)
2020-01-10 16:40:32,929 openrave [WARN] [fclspace.h:248 FCLSpace::KinBodyInfoPtr fclrave::FCLSpace::InitKinBody] Initializing link teoSim/LeftArmTCP with 0 geometries (env 1) (userdatakey fclcollision0x55ec3464df60)
2020-01-10 16:40:32,943 openrave [WARN] [fclspace.h:248 FCLSpace::KinBodyInfoPtr fclrave::FCLSpace::InitKinBody] Initializing link teoSim/r5dummy with 0 geometries (env 1) (userdatakey fclcollision0x55ec3464df60)
2020-01-10 16:40:32,963 openrave [WARN] [fclspace.h:248 FCLSpace::KinBodyInfoPtr fclrave::FCLSpace::InitKinBody] Initializing link teoSim/RightLegTCP with 0 geometries (env 1) (userdatakey fclcollision0x55ec3464df60)
2020-01-10 16:40:32,975 openrave [WARN] [fclspace.h:248 FCLSpace::KinBodyInfoPtr fclrave::FCLSpace::InitKinBody] Initializing link teoSim/r6dummy with 0 geometries (env 1) (userdatakey fclcollision0x55ec3464df60)
2020-01-10 16:40:33,000 openrave [WARN] [fclspace.h:248 FCLSpace::KinBodyInfoPtr fclrave::FCLSpace::InitKinBody] Initializing link teoSim/LeftLegTCP with 0 geometries (env 1) (userdatakey fclcollision0x55ec3464df60)
2020-01-10 16:40:33,088 openrave [WARN] [fclspace.h:248 FCLSpace::KinBodyInfoPtr fclrave::FCLSpace::InitKinBody] Initializing link teoSim/FrontalRightWrist with 0 geometries (env 1) (userdatakey fclcollision0x55ec3464df60)
2020-01-10 16:40:33,088 openrave [WARN] [fclspace.h:248 FCLSpace::KinBodyInfoPtr fclrave::FCLSpace::InitKinBody] Initializing link teoSim/RightArmTCP with 0 geometries (env 1) (userdatakey fclcollision0x55ec3464df60)
2020-01-10 16:40:33,106 openrave [WARN] [fclspace.h:248 FCLSpace::KinBodyInfoPtr fclrave::FCLSpace::InitKinBody] Initializing link teoSim/FrontalLeftWrist with 0 geometries (env 1) (userdatakey fclcollision0x55ec3464df60)
2020-01-10 16:40:33,106 openrave [WARN] [fclspace.h:248 FCLSpace::KinBodyInfoPtr fclrave::FCLSpace::InitKinBody] Initializing link teoSim/LeftArmTCP with 0 geometries (env 1) (userdatakey fclcollision0x55ec3464df60)
2020-01-10 16:40:33,117 openrave [WARN] [fclspace.h:248 FCLSpace::KinBodyInfoPtr fclrave::FCLSpace::InitKinBody] Initializing link teoSim/r5dummy with 0 geometries (env 1) (userdatakey fclcollision0x55ec3464df60)
2020-01-10 16:40:33,137 openrave [WARN] [fclspace.h:248 FCLSpace::KinBodyInfoPtr fclrave::FCLSpace::InitKinBody] Initializing link teoSim/RightLegTCP with 0 geometries (env 1) (userdatakey fclcollision0x55ec3464df60)
2020-01-10 16:40:33,149 openrave [WARN] [fclspace.h:248 FCLSpace::KinBodyInfoPtr fclrave::FCLSpace::InitKinBody] Initializing link teoSim/r6dummy with 0 geometries (env 1) (userdatakey fclcollision0x55ec3464df60)
2020-01-10 16:40:33,171 openrave [WARN] [fclspace.h:248 FCLSpace::KinBodyInfoPtr fclrave::FCLSpace::InitKinBody] Initializing link teoSim/LeftLegTCP with 0 geometries (env 1) (userdatakey fclcollision0x55ec3464df60)
2020-01-10 16:40:33,215 openrave [WARN] [environment-core.h:492 Load] load failed on file /usr/local/share/xgnitive/contexts/models/teo_cgda_iros.env.xml
[info] OpenraveYarpPluginLoader.cpp:22 OpenraveYarpPluginLoader(): Checking for yarp network...
[success] OpenraveYarpPluginLoader.cpp:25 OpenraveYarpPluginLoader(): Found yarp network.
yarp: Port /OpenraveYarpPluginLoader/rpc:s active at tcp://10.118.40.231:10002/
yarp: Port /OpenraveYarpPluginLoader/state:o active at tcp://10.118.40.231:10003/
[info] OpenraveYarpPluginLoader.cpp:168 Open(): Checking for yarp network...
[success] OpenraveYarpPluginLoader.cpp:175 Open(): Found yarp network.
[debug] OpenraveYarpPluginLoader.cpp:182 Open(): config: (device controlboardwrapper2) (genRefSpeed 999999) (manipulatorIndex 2) (robotIndex 0) (subdevice YarpOpenraveControlboard)
[info] OpenraveYarpPluginLoader.cpp:185 Open(): penv: 0x55ec347d0840
[DEBUG]ControlBoardWrapper2: 'period' parameter missing, using default thread period = 20ms 
[DEBUG]opening controlBoardWrapper2 subdevice
[debug] DeviceDriverImpl.cpp:27 open(): config: (device YarpOpenraveControlboard) (genRefSpeed 999999) (manipulatorIndex 2) (name "/teoSim/rightLacqueyFetch/internalFinger") (penv {64 8 125 52 236 85 0 0 224 241 122 52 236 85 0 0}) (robotIndex 0) (subdevice YarpOpenraveControlboard)
[debug] DeviceDriverImpl.cpp:64 open(): Get JointPtr for manipulatorIDs[0]: 38 (rightLacqueyFetch/InternalFingerProximal)
[debug] DeviceDriverImpl.cpp:64 open(): Get JointPtr for manipulatorIDs[1]: 39 (rightLacqueyFetch/InternalFingerDistal)
[debug] DeviceDriverImpl.cpp:80 open(): pcontrol: 0x55ec35d634e0, IdealController
[info] DeviceDriverImpl.cpp:86 open(): Detected idealcontroller, switch to genericmulticontroller.
[debug] DeviceDriverImpl.cpp:99 open(): pcontrol: 0x55ec34851330, genericmulticontroller
[debug] DeviceDriverImpl.cpp:116 open(): Attach individual controller for manipulatorIDs[0]: 38
[debug] DeviceDriverImpl.cpp:116 open(): Attach individual controller for manipulatorIDs[1]: 39
[INFO]created device <YarpOpenraveControlboard>. See C++ class roboticslab::YarpOpenraveControlboard for documentation.
[info] IPositionControlImpl.cpp:15 getAxes(): Reporting 2 axes are present
[DEBUG]joints parameter is 2
[info] IPositionControlImpl.cpp:15 getAxes(): Reporting 2 axes are present
[info] IPositionControlImpl.cpp:15 getAxes(): Reporting 2 axes are present
[INFO]/teoSim/rightLacqueyFetch/internalFinger : no ROS initialization required 
[INFO]/teoSim/rightLacqueyFetch/internalFinger  initting YARP initialization 
yarp: Port /teoSim/rightLacqueyFetch/internalFinger/rpc:i active at tcp://10.118.40.231:10004/
[... cut ...]
[info] OpenraveYarpPaintSquares.cpp:184 Open(): port name: /openraveYarpPaintSquares/rpc:s
[info] OpenraveYarpPaintSquares.cpp:187 Open(): squares: 64
2020-01-10 16:40:33,376 openrave [INFO] [OpenraveYarpPaintSquares.cpp:189 Open] penv: 0x55ec347d0840
sucess: object "object" exists.
sucess: object "wall" exists.
sucess: object "palete-magenta" exists.
sucess: object "palete-yellow" exists.
sucess: object "palete-cyan" exists.
Robot 0: teoSim
yarp: Port /openraveYarpPaintSquares/rpc:s active at tcp://10.118.40.231:10008/
El color con el que estoy pintando es: 1
```

jgvictores, 2020-01-10:
Yes, this is due to missing https://github.com/roboticslab-uc3m/forceSensor (a fork off https://github.com/robEllenberg/forceSensor). It's not documented because installation is not perfect, so I wanted to merge it into this repo to take advantage of the existing installation mechanisms.

RaulFdzbis, 2020-01-10:
But I think I have forceSensor installed in my computer, this is the output of `locate libForceSensor`:

```bash
/home/raul/repos/forceSensor/lib/libForceSensor.so
/home/raul/repos/plugins/libForceSensor.so
/usr/local/lib/openrave0.9-plugins/libForceSensor.so

```

jgvictores, 2020-01-10:
`sudo /home/raul/repos/plugins/libForceSensor.so /usr/local/lib/openrave0.9-plugins/libForceSensor.so
`

RaulFdzbis, 2020-01-11:
Yep, copying the file solved the problem. 

It looks like i have forgotten to do a `sudo updatedb` before the locate so the output i put was outdated.

Thanks!, I think we can close this :D

closed 2020-01-11 by RaulFdzbis