roboticslab-uc3m/openrave-yarp-plugins #118: Correctly simulate speeds and accelerations

rsantos88, 2020-05-29:
Trying to simulate the presentation demo, as well as other scripts that require modifying the standard speed and acceleration, I've found that these warning messages appear, indicating that it exceeds the speed of 10:
```bash
[warning] IPositionControlImpl.cpp:196 setRefSpeed(): Setting 25,000000 refSpeed above 10,000000 maxVelLimit. All joint 5 movements will be immediate.
```
The behavior is of immediate displacements (the message indicates this).
How can this limit be configured? and where can I also configure the acceleration limit?
Attached video (zip compressed): [teo-sim-warning.zip](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/files/4699657/teo-sim-warning.zip)
Thanks!!

PeterBowman, 2020-05-29:
Velocity limits need to be raised a bit. This is easier achieved through the [`IControlLimits` interface](http://www.yarp.it/classyarp_1_1dev_1_1IControlLimits.html): either add the convenient method calls in your code (which means hardcoding), or adjust these values prior to running the task via RPC calls (preferably). Getters are exposed [in this guide](http://robots.uc3m.es/gitbook-developer-manual/appendix/yarp-tricks.html#limits), setters are easy to infer (`set vlim 0 value`).

Alternatively, change the max velocities directly in the OpenRAVE model files ([example](https://github.com/roboticslab-uc3m/teo-openrave-models/blob/95dfc4b9137ee9b6c9a97d32b87951de7c1d0bc7/openrave/teo/teo_leftArm.kinbody.xml#L27)).

There are no acceleration limits, just position and velocity.

jgvictores, 2020-06-01:
As a historical note, this feature is preserved due to backward compatibility with old Xgnitive experiments, which should actually be ported to use the `IPositionDirect` interface (not even sure if this has been done already).

@rsantos88 are the proposals above sufficient for the time being?

rsantos88, 2020-06-01:
> ...or adjust these values prior to running the task via RPC calls (preferably). Getters are exposed in this guide, setters are easy to infer (set vlim 0 value).

After doing some tests with this option, comment that to configure it by RPC correctly it would be `set vlim njoint minvel maxvel` and it works great. For example: `set vlim 0 -20 20`
Anyway I want to try the other options later.

> @rsantos88 are the proposals above sufficient for the time being?

Yes. Thanks for your help. I'll close this issue :+1:

closed 2020-06-01 by rsantos88